pavel-demin/red-pitaya-notes #772: Use the VNA app for output power measurement not just Gain

satfan52, 2018-10-29:
Hello Pavel, 

I am an intensive user of the VNA. It is a great success, thank you very much for your efforts. 

At the moment I am using it to measure the Gain of my home made rf power amifier (with adequate attenuators in the input and output path of course) 

To measure the rf output power accurately, I can t use the VNA app so I use rf power meters based on AD8307 and 8317 chips. The procedure is accurate but quite tedious as it requires extensive calibration and manual operation to measure accurately the output power of my home made rf amplifier over the whole HF range. I was wondering if instead I could use / modify the VNA client in order to automate the measurement of the output power of my LDMOS amplier in one single sweep between 1 and 54 Mhz (similar to gain short  measurement). Of course I assume that some calibration would also be required but instead of doing it manually  in Excell as I do it at the moment with my AD8307 and 8317 chips, maybe it could be done by the application based on a "reference power level" at several frequencies. 

Is this worthwhile investigating? 

Regards
Peter

pavel-demin, 2018-10-29:
Hello Peter,

I'm glad that the VNA works well for you.

I don't have any idea of how to measure the output power of an amplifier. Could you please provide more details about what exactly should be measured?

I've just had a look at the following two documents:
https://www.keysight.com/upload/cmc_upload/All/BTB_PowerBasics_2005.pdf
https://www.boonton.com/~/media/Boonton/Reference%20Guides/WTG_RefGuide_F1128_sm_web.pdf

They both mention power sensors that convert power to DC voltage. Are you using something similar?

Best regards,

Pavel

satfan52, 2018-10-29:
Just to clarify what I am measuring, it is actually not just the final stage (so a PA), but a pre-driver (MHW592), a driver (pushpull MRF134) and the final stage (pushpull VRF 151). So a whole amplifying chain.

I use two of these power meter devices:

https://www.ebay.com/itm/1-500Mhz-OLED-RF-Power-Meter-70-15dBm-1nW-2W-Power-Set-RF-Attenuation-Value/222156133908?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

This rf power meters are based on the famous AD8307 chip of Analogue Devices which is quite accurate between 0 and 60 Mhz. I verified that the rms voltage indicated by this rf power meter matches the one measured done with the red-pitaya oscilloscope as well as with another oscilloscope. That's how I am confident that the power indication of these device is OK over a wide dynamic range as long as they are being used well below their 1dB compression point (so roughly around 0 dBm). The dynamic range of these AD8307 based devices is 70dB which is not bad at all.   
  
One of the two rf power meter is placed at the input of the amplifier and the other one at the output of the amplifier as I want to measure input and output and gain at the same time. I use adequate attenuators and power-bridges on the input and output of the amplifier 
- so I don't grill the rf power meter in the output path, 
-  to ensure that power measured with these devices are always over perfect 50 ohm load,
- to ensure that power bridges used see 50 ohms on all their ports (except on the port of the signal generator)
-   to ensure that the power of the signals measured is in the range of -20-60dBm so far below their 1dB compression points around 0 dBm.

This set-up  allows to measure, the input power and the output power as well as the gain at the same time (by deducting the difference between the power indicated on the input and output rf power meters). Fortunately, the gain measured that way  matches the one measured by the VNA, quite reasuring ;-) ;-).  

I do all the math in Excell to compensate for the power bridge and attenuator effects. I verify that the set-up and my Excell calculations is correct is by measuring a reference attenuator. If the attenuation measured that way matches the attenuation measured with the VNA app, then my set-up is fine. 

The procedure is quite tedious because of the need to use Excell to do the math, the need to set manually the frequency of  measurement on the two rf power meters (there is some frequency compensation build in the firmware of these devices, so settings must be changed manually for each frequency on each of the power meters).

Ideally I would like to be able to measure the gain, and the corresponding input and output power at the same time with the VNA app all in one sweep (and not the gain only). I was thinking that maybe the "DUT tab" or a "new tab" could possibly indicate the power input and output power (in dBm ? or dBFs ?) as measured during gain-short sweeps ?  Doing a "gain-short" sweep would then provide the gain (in dB) under the gain-short tab (as it is today) as well as input and output power (in dBm? or dBFS ?) under the modified DUT tab or a under new one . 

Example:  
- measured input power from -15 and -17 dBm   
- measured output power  from  -35  to -30 dBm 
- measured gain -20 to -13 dB
If I have a 85 dB attenuator on the output path and zero attenuator in the input path I could then easily derive that 
- input power varies between -15 and -17 dBm (since no attenuator in the input path )
- output power varies between 50dbm (100W) and 55 dBm (about 300W)  (since 85 dB att in the output path) 
- total gain varies between 65 dB and 72 dB (since 85 dB att in the ouput path)

By the way  I tried a very noce classical ham SWR/Power meter ( Daiwa CN901-HP3.) It is nice for everyday operation to get an indication of the output power, it is even accurate at "some" frequencies and "some" power levels,  but not good when trying to make accurate max power measurements from 2 to 54 Mhz.

pavel-demin, 2018-10-29:
Thanks a lot for the details. I think that I'm starting to understand how the power measurement works.

So, if I'm not mistaken, AD8307 is the power sensor and it converts RF power to DC voltage. Then the DC  voltage should be measured with an ADC.

The measurement process could look like the following:
1. generate a fixed frequency signal (for example, using Red Pitaya RF outputs)
2. measure the DC voltages from the power sensors (for example, using the Red Pitaya RF inputs)
3. increase the frequency and go to step 1

Is my understanding of the process correct so far?

I don't think that the VNA application would work for this measurement process because the input ADC samples are multiplied by cos and -sin:
http://pavel-demin.github.io/red-pitaya-notes/img/vna.png

I think that this kind of measurement requires a separate application.

satfan52, 2018-10-29:
> Thanks a lot for the details. I think that I'm starting to understand how the power measurement works.
> So, if I'm not mistaken, AD8307 is the power sensor and it converts RF power to DC voltage. Then the DC  voltage should be measured with an ADC.

That is correct.  It's a logarithmic detector by the way not a linear one. The conversion V /dBm is based on a conversion table provided by Analogue device. 

> The measurement process could look like the following:
> 
> generate a fixed frequency signal (for example, using Red Pitaya RF outputs)
> measure the DC voltages from the power sensors (for example, using the Red Pitaya RF inputs)
> increase the frequency and go to step 1
> 
> Is my understanding of the process correct so far?

Yes correct, the minimum requirement would be to measure the output power. Ideally both input and output voltages (so the gain also comes as an additional benefit). If attenuators are used in the input and output, it would be great to enter the value so the result displayed is the actual true input and output power (and gain)

> I don't think that the VNA application would work for this measurement process because the input ADC samples are multiplied by cos and -sin:

But the DUT data is actually the raw output data ? it is not possible to derive a power indication from it ?

> I think that this kind of measurement requires a separate application.

Unfortunately

pavel-demin, 2018-10-29:
> But the DUT data is actually the raw output data ?

The DUT data is the ADC samples multiplied by cos and -sin, filtered and down-sampled by 2500.

> it is not possible to derive a power indication from it ?

I'd say that it's not possible but maybe I'm mistaken.

> Unfortunately

I'm nor sure why. The unmodified VNA application can't be used for the power measurements. Some modifications of the FPGA configuration and of the user interface are required. I think that it would be easier to create a new power measurement application with some parts borrowed from the VNA application than to try merge both applications into a single one.

satfan52, 2018-11-03:
Actually this application would have to filter out all harmonics to ensure that during the sweep  only the power of the target fundamental frequency  is measured and that the power of its harmonics are not included in the measurement. That is why I also use the spectrum analyzer in the output path when I want to make  accurate manual power measurements  (their effect of harmonics is not negligible). As you can imagine the whole manual procedure, to measure accurately the output power  of the entire HF band takes time and is quite tedious and it is easy to make some errors. A simple app (with the input and output attenuator as GUI parameter) that would in one single sweep report input power (in dBm), output power (ind dBm) and gain (in DB) (taking into account the value of input and output attenuators of course) would be fantastic and  great time saver. If in in addition, this app could also generate a two tone signals (with the power of each two tone signal set exactly 6dB below the power of the input signal used for the power measurement) in order to measure the IM3  in one single sweep, it would be another miracle.

pavel-demin, 2018-11-03:
> Actually this application would have to filter out all harmonics to ensure that during the sweep only the power of the target fundamental frequency is measured and that the power of its harmonics are not included in the measurement.

I'm afraid that I don't understand this part. Here are my questions:
 - If the power sensor's output is a DC voltage, then how to filter it?
 - On page 18 in [WTG_RefGuide_F1128_sm_web.pdf](https://www.boonton.com/~/media/Boonton/Reference%20Guides/WTG_RefGuide_F1128_sm_web.pdf), I read the following:

> This narrowband, or tuned measurement method is quite different from a wideband sensor-based power measurement, and the reported result will often not agree with a true power measurement.

So, I don't understand what kind of power measurement are you looking to implement, narrowband or wideband. After reading the linked document, I have an impression that the true power measurements should be wideband.

If the narrowband measurement is OK for you, then maybe the receiver-based amplitude measurement as described on page 18 in [WTG_RefGuide_F1128_sm_web.pdf](https://www.boonton.com/~/media/Boonton/Reference%20Guides/WTG_RefGuide_F1128_sm_web.pdf) is the way to go. It can be done without any power sensors, logarithmic detectors, etc. Am I missing anything?

satfan52, 2018-11-04:
Dear Pavel,

Of course you are not missing anything. I am sorry if I was not clear enough.

Just to avoid any confusions, please  note that that I am measuring power of sine-waves (CW) not of modulated signals. That's an important point not to be missed in the discussion between narrow band and wide band power measurement methods.

I made my first power measurements of the output of my homemade amplifier using "wide band power sensors" based on AD8307/8317 chips (that convert power into voltage), but I found the results of my measurements to be less accurate then when I used the redpitaya spectrum analyser. The  spectrum analyser app allows to measure the power of the fundamental signals and of some of its harmonics which is something that the wide band power detector method is unable to do for obvious reasons (as you pointed out yourself and as described very well in the paper that you referred to). 

I am using both methods  but my favourite one is clearly the narrow bandwidth one using a spectrum analyser. I find  the use of the spectrum analyzer the most accurate method to characterize an rf amplifier when the input signal is a CW sine. I f my understanding is right, the paper you referred to reaches the same conclusion.

My point is that the narrow bandwidth or wide bandwidth methods are slow, tedious and error prone and that it would be nice to have an app that automates it to be able to measure the performance of wideband HF amplifiers in a fast and easy way. My idea is to connect the redpitaya output 1  at the input if the amplifier via an input attenuator, connect the redpitaya input 1 at the output of the rf amplifier via an output attenuator, click on some button and get in return in one single frequency sweep all the info that I mentioned earlier (see the list in my previous post). Obviously to get info about the level of the second and third harmonics,  and the IMD3 only the narrow bandwidth method based on some form of spectral analysis is relevant.   At the moment I am doing all these repetitive measurements (narrow or wide bandwidth ones) at 2, 3, 4, 5, 7, 10, 14, 18, 21, 28, 30, 50 and 52 Mhz and believe me it takes me time and energy as everytime I want to change something in my rf amplifier to try to optimize it (for instance the output TLT, the bifillar choke, the gate source feedback network or simply the bias current), I have to repeat all my measurements to see the effect of the changes I introduced at different frequencies. If I had an app, to measure input power, output power, gain , level of 2nd and 3rd harmonics, IMD3, gain compression in one go between 0 and 60 Mhz, it would be much simpler to experiment with rf amplifiers, that is all I am saying.

Thank you again for your great work

Regards
Peter

pavel-demin, 2018-11-06:
Have you found a solution?

Otherwise, let's keep this issue open and continue to accumulate the information about power measurements.

Once we have a clear understanding of how exactly this kind of measurements could be implemented on Read Pitaya, it'll be a very good starting point for a new project.

satfan52, 2018-11-08:
Hello Pavel,

No of course, no solution yet. I didn't find an app that would automate all these gain and power measurements.

a) I have however tried a third method, the QUISK method. I use the QUISK and the HPSDR transceiver to measure the signal strength with the QUISK S meter (in dBFS). 

b) The dynamic range in this set up with the  50KSA sampling rate is quite good and exceeds the one of the AD8307 by far (thanks to the decimation).  The reliable dynamic range of the AD8307 is  about 70dB according to the specifications and this is what I have observed as well, during my experimentations. The dynamic range of the AD8307  is of course well inferior to the red pitaya QUISK/HPSDR txcv set-up one assuming a sampling rate of 50 KSA. Signal to noise ratio is just better than with an AD8307 that just converts power into voltage, so inevitably the power measurement is more accurate as well. 

c) I am must say I am impressed with he quality of the QUISK S meter which seems relatively accurate as long as the signal measured is far below the  -1 dB commpression point and as long as the signal to noise ratio is good (well above the noise floor and measured in adequate bandwidth).  So the trick I use to make reliable and accurate power measurement is to calibrate my measurements in dBFS based on the measurement of the AD8307 in dBm. Once the "GAIN" (actually the transfer function) of the QUISK set-up is known, it is possible to use QUISK to make accurate power measurements with the same if not better accuracy than the AD8307. The advantage of using QUISK over the AD8307 are multiple, it is not just the dynamic range, it is also the possibility of measuring harmonics as well as the IMD3 (during a two tone test).  These things are not possible with the AD8307 of course, but the AD8307 is useful to calibrate QUISK. This way, I actually get the best of both worlds, the accuracy of the AD8307 with the better linearity and dynamic range of the redpitaya/hpsdr txcv/quisk app (cfr the paper you refered too).  By the way the AD8317 is not as good as the AD8307 for HF power measurement, it is designed principally for power measurement in the GHz range and the dynamic range is much lesser too (50dB)

d) The same approach can be followed with the spectrum analyser app, but the approach is not as performant as with the QUISK set-up; The spectrum analyser app stops measuring peaks automatically above 50 Mhz (out of range notification) . The spectrum analyser app also has a lesser dynamic  range and a lesser frequency resolution that prevent of measuring IMD3 of two tone test signals which are only distant by a few Khz. The only advantage of the spectrum analyzer app over the Quisk set-up is that it is possible to view the whole spectrum at once and check in a single glance all the harmonic. Whereas in the QUISK set-up that's not an option, each harmonic must be tuned at sperately to be measured  (If you remember we had a discussion about the bandscope functionality of QUISK which is not available as that supporting part of the HPSDR protocol is not implemented).  Ideally, for power measurement, you want the accuracy, frequency resolution and the dynamic range of a QUISK like set-up with the wideband panoramic capabilities of the spectrum analyser like app.      

e) Again I think the key challenge is to succeed to automate all these power measurements as most of the time, one does not wants to measure only the fundamental signal at one single frequency, but also the harmonics (and maybe also unwanted Aliases possibly unwillingly amplified) for SEVERAL frequencies. Using the quisk/hpsdr/redpitaya method with some manual calibration on an AD8307 is really slow, but I agree it is better than nothing.

Regards
Peter

satfan52, 2018-11-09:
some starting ideas for high level specification of a new app 

[ideas for a high level spec fo a new SNA & Power measurement app.docx](https://github.com/pavel-demin/red-pitaya-notes/files/2566841/ideas.for.a.high.level.spec.fo.a.new.SNA.Power.measurement.app.docx)

mgarrett1955, 2024-10-21:
On the subject of RF power ammmplier measurement, we plan to sample voltage and current output of an amplifier, and calculate the voltage, current, phase and power factor. In order to do this, we need to capture the two channesl and have them be phase coherent. We tried the gnuradio am_tranciever.grc, and added the second channel using another source set to sockets 1001, 1002 and noted that the sampled intpu data, does not seem to be phase coherent. Is there a mechanims in the Red pitaya to get two channels phase coherent into gnuradio? Thank you in advance!!