pavel-demin/red-pitaya-notes #774: Frequency sweep meter

sq3rx, 2018-11-04:
Hi Pavel,
First of all thanks for all apps which you develope for RP - all of them all very useful for me.
I would like to suggest you to add one more simple app to your collection, which can be very useful - frequency sweep analyzer. It allows measure insertion loss, cable loss etc. I think it could be possible to develop this based on VNA app. 

You can take a look for eg.  NWT70, NWT7, NWT500 equipment which can provide this kind of measurements.

satfan52, 2018-11-04:
Actually the current VNA can do this. To calibrate just connect input 1 and output 1 via two attenuators (I use a 10dB for input and a 20db one for output) and hit the "short button", this will do the  "through" calibration. Then insert the "DUT" between the two attenuators and hit the "single" button. Check results of the sweep under the "gain short"  tab. I use it all the time.

pavel-demin, 2018-11-06:
I've just checked the NWT70 product description at [this link](http://www.giga.co.za/ocart/index.php?route=product/product&product_id=226) but I can't understand from the description what exactly it does.

If it measures gain vs frequency , then the method described by Peter is exactly how this measurement can be done using the VNA application.

sq3rx, 2018-11-06:
Generaly this equipment measure frequency vs gain. Thanks for detailed description how to measure this by vna app. I will test it probably during the weekend.

pavel-demin, 2018-11-07:
> I would find such an instrument very useful.

Have you tried the VNA application?

> While the existing frequency response instrument is extremely useful, a swept frequency instrument, where the sweep rate could be slowed down for narrower noise banding to give a better signal to noise, would be good.

The VNA application supports bandwidths from 1 Hz to 10 kHz.

> But if someone else is planning to do it, I have plenty else to do !

I think that it's already done. The VNA application can be used to measure gain vs frequency with variable bandwidth.

satfan52, 2018-11-08:
The only additional thing that the NWT4000/5000/6000 series can do compared to the red-pitaya VNA is spectrum analysis between 25 and about 6Ghz (for the NWT6000). These devices are actually spectrum analysers and include  tracking generators in order to make up a scalar vector analyser (ie, frequency response analyser). They use ADF4351 type of devices in combination with AD8307/17 type of logarithmic detector devices. So for me, unless I am mistaken, the redpitaya VNA is actually providing the same "scalar network analysis" than the NWT4000/5000/6000 series but between 0 and 60 mhz only whereas the NWTs family start in the MHz ranhe and climbs in the Ghz range. However the RP does it much better in its frequency range than NWTs because of the greater dynamic range of the RP compared to the logarithmic detectors used in these NWT devices.  Personally, I think the added value for the red-pitaya VNA would be to add some kind of frequency converter to be able to use the redpitaya well above 60 Mhz with a similar quality of measurement than below 60 Mhz; image rejection dynamic range would be the issue to tackle. This is the approach that Mini Radio Solution took with the frequency extender to their mini VNA pro (not sure how they did though, if they used aliaising and under sampling or a true old fashion analogue up converter)  that extends the measuring frequency from 200 Mhz up to about 1.5 Ghz. Of course some changes would be required in the VNA app as well to support such an up converter/ frequency extendedr.  There are other examples of HAM VNAs using frequency extenders, and unless I am mistaken even HP boat anchors from the 80s used some frequency extenders. For me this is objectively what is missing to the red-pitaya VNA today so it can used in a similar way to  NWT4000/5000/6000 series, is simply some frequency extension above 60 Mhz. Of course, the added value is the V of VNA, it would be extremely useful to measure impedances and phase above 60 Mhz as well for designing filters etc... provided it can be done sufficiently accurately and with a reasonable dynamic range.