pavel-demin/red-pitaya-notes #776: RAM writer in the ADC projects

klujby, 2018-11-13:
Hi Pavel,

Nice work with the ADC projects, I managed to modify the adc_recorder (based on comments in other threads) and save up to 64M sample pairs.

I am now trying to implement continuous & lossless data recording, and I would like to gain some insight about the axis_ram_writer core.
My question is: What happens if the writer gets to the end of the available memory space in 'adc_recorder' and 'adc_test' projects? Does it start from the beginning automatically again or does it wait for a reset? 
I am thinking about having a writer that acts like a ring by automatically resetting itself, and stops if the writer position (which, if I understand correctly, is sts_data) reaches a certain value (that may be set from a c program as well). Do you think it is possible to create one like that?

Also, I am very new to fpga programming and I am having quite a lot of trouble understanding the syntax and functionality of the core and block design scripts. Would you reference some materials that can help me getting better at this topic?

Best regards,
Gergely Klujber

klujby, 2018-11-14:
Hi Pavel,

Thank you for the quick and detailed answer, and the reference material.

> It's exactly how it works in the adc_recorder_trigger and mcpha projects

That is perfect, I will look into those projects as well.

liubenyuan, 2018-11-16:
Thanks! This is indeed helpful.

closed 2018-12-06 by pavel-demin