pavel-demin/red-pitaya-notes #778: End of WSPR transmission detection

davidhay45, 2018-11-16:
This is a question.

Using transmit-wspr-message from the SD card 3.8.

The transmission of a wspr message is initiated by memory mapped i/o:

   *fifo = (uint32_t)floor((1.0 + 1.0e-6 * corr) * dphi * (1<<30) + 0.5);

and the process then runs independently for slightly under two minutes within the Redpitaya processor.

Is there a way to detect that the transmission has finished?  This is to allow control of an rf amplifier and low pass filter settings.

Thanks

pavel-demin, 2018-11-16:
> Is there a way to detect that the transmission has finished?

Since we know when the transmission starts and we know its duration, I think that the time when the transmission ends can be easily calculated.

> This is to allow control of an rf amplifier and low pass filter settings.

The commands controlling an RF amplifier can be found [on line 31 in decode-wspr.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/decode-wspr.sh#L31) and [on line 22 in transmit-wspr.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/transmit-wspr.sh#L22).

The idea is to change the state of an RF amplifier before the transmission starts and before the reception starts.

By default, these commands control the pin DIO0_N of the [extension connector E1](http://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extent.html#extension-connector-e1). They can be easily modified to control other pins (DIO1_N - DIO7_N).

It's also possible to add commands that would control filters via other onboard interfaces (I2C, SPI, USB, Ethernet, etc).

davidhay45, 2018-11-20:
I have a band pass filter, a low pass filter and a switchable pre-amp and all are controlled by D100-P - D107_P along with the PTT of my linear amp.  What I was asking for was a positive detection of end-of-transmission rather than a timing loop so that I could move on to the next band at the next even minute without having to miss a 2 minute period.

That appears not to be possible but thanks for looking at it.

closed 2018-11-20 by davidhay45

pavel-demin, 2018-11-20:
I'm still not sure if I understand the problem that you're trying to solve.

If you want to execute a command just after the transmission, then you can add it at the end of the [transmit-wspr.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_wspr/app/transmit-wspr.sh) script:
```
$TRANSMITTER $CONFIG

# wait 111 seconds
sleep 111

# the following command runs after the transmission
<command>
```

The duration of the transmission is 110.6 seconds. So, the command after `sleep 111` will be executed 400 ms after the end of the transmission.

davidhay45, 2018-11-21:
I don’t like the idea of sleep 111 because I can’t monitor other things at the same time.  I suppose I am a purist but I am used o positive signals of events rather than timing.  It comes form years of designing software to control telephone exchanges and other process control systems.

Thanks for your help, as always.

From: Pavel Demin 
Sent: Tuesday, November 20, 2018 21:49
To: pavel-demin/red-pitaya-notes 
Cc: davidhay45 ; State change 
Subject: Re: [pavel-demin/red-pitaya-notes] End of WSPR transmission detection (#778)

I'm still not sure if I understand the problem that you're trying to solve.

If you want to execute a command just after the transmission, then you can add it at the end of the transmit-wspr.sh script:

$TRANSMITTER $CONFIG

# wait 111 seconds
sleep 111

# the following command runs after the transmission
<command>
The duration of the transmission is 110.6 seconds. So, the command after sleep 111 will be executed 400 ms after the end of the transmission.