pavel-demin/red-pitaya-notes #780: question: software compatibility with the new Hardware RP announced on 22 November ?

satfan52, 2018-11-27:
Hello Pavel,

Probably too early to say (??) , but will your different software remain compatible with the new RP announced on 22 November 2018 ? 
see: https://www.redpitaya.com/n86/new-stemlab-sdr
I am especially thinking of the VNA application and the HPSDR compatible transceiver.
Is there something to gain for the end-user in terms of new or enhanced software functionalities (other than better performance) from the new hardware  ?

Thanks
Regards
Peter

pavel-demin, 2018-11-27:
I'm planning to port the applications to the new board.

I've created a repository branch for the ported applications:

https://github.com/pavel-demin/red-pitaya-notes/tree/122-16

The 122.88 MHz oscillator on the new board allows to implement the wide-band frequency display in the HPSDR software (https://github.com/pavel-demin/red-pitaya-notes/issues/673).

bjornekelund, 2018-11-30:
This is great news. Do you expect 7020 FPGA with three times as many logic elements as the current 7010 to allow for more than 8 receivers in your multi-band receiver application? It would be amazing to have a single RP act as skimmer for all ham bands 160-6m including 60m.

pavel-demin, 2018-11-30:
While Zynq-7020 has enough FPGA resources for 16 (or even more) receivers, the number of CPUs is still the same. It's quite possible, that the CPU resources are not sufficient to decode the WSPR and FT8 signals from 16 bands.

bjornekelund, 2018-11-30:
I understand, but I'm mainly interested in running CW Skimmer Server which leaves all the decoding to the host PC. My current PC runs at 8-12% load with 8 bands so it should be feasible from that perspective. To run a 16 band CW skimmer would be fantastic. Even 12 bands would be great.

bjornekelund, 2018-11-30:
Thanks. I see. How unfortunate.

pavel-demin, 2018-11-30:
It's possible to run two instances of CW Skimmer Server with two 125-14 Red Pitaya boards.

If you already have one 125-14 Red Pitaya board, then it's cheaper to buy an additional 125-14 Red Pitaya than to replace the 125-14 board with a 122-16 one.

bjornekelund, 2018-11-30:
Thanks. I'm well aware of this and I have successfully tested it. Since the inputs are high impedance it is also easy to share the antenna signal after the step-up transformer. I would however have hoped to switch to just a single one with better receiver performance, 50 ohm input etc.

satfan52, 2018-11-30:
Yes Gfeat news indeed. On the topic of QUISK compatibility (and I guess PowerSDR too) it is really  great that the hpsdr sever will allow the support the band scope functionality

pavel-demin, 2019-02-24:
I've created a separate GitHub repository and a separate set of notes on the new STEMlab 122.88-16 SDR board. Here are the links:

https://github.com/pavel-demin/stemlab-sdr-notes
https://pavel-demin.github.io/stemlab-sdr-notes

closed 2019-02-24 by pavel-demin