pavel-demin/red-pitaya-notes #781: Question: Using a second HP Slave AXI Interface

blowfish2003, 2018-11-27:
Hello Pavel,

I am new to RP and FPGA programming and your tutorial and work have been immensely helpful. Thank you very much for all the work you have been doing.

I am currently working on a model to sample signal changes on GPIO pins and store the number of clock cycles as well as a counter snapshot when change occurs at two different DDR memory locations. For these I am following your adc_recorder projects.

In these projects you have used certain setting for setting HP interfaces:

`  ram = mmap(NULL, 8192*sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0x1E000000);

  /* set HP0 bus width to 64 bits */
  slcr[2] = 0xDF0D;
  slcr[144] = 0;
  axi_hp0[0] &= ~1;
  axi_hp0[5] &= ~1;
`
Can you please tell me how to adapt these settings in case I want to use two HP interfaces simultaneously?

Thanks in advance for your help.

blowfish2003, 2018-11-28:
Hi, 

I have tried to instantiate the HP2 in the following manner:

` 
 axi_hp0 = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, mmapfd, 0xF8008000);
  axi_hp2 = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, mmapfd, 0xF800A000);
  sts = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, mmapfd, 0x40001000);
  sts_c = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, mmapfd, 0x40002000);
  ram = mmap(NULL, 2048*sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, mmapfd, 0x1E000000);
  ram_c = mmap(NULL, 2048*sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, mmapfd, 0x1EFF0000);

  /* set HP0 bus width to 64 bits */
  slcr[2] = 0xDF0D;
  slcr[144] = 0;
  axi_hp0[0] &= ~1;
  axi_hp0[5] &= ~1;

  /* set HP2 bus width to 64 bits */
  axi_hp2[0] &= ~1;
  axi_hp2[5] &= ~1;
`
But I am not reading any data on the buffers. Can you please tell me if I am using it wrong.

Thanks.

pavel-demin, 2018-11-28:
I've added the code setting the HP0 bus width to 64 as a workaround for the problem discussed in [this issue](https://github.com/RedPitaya/RedPitaya/issues/89). The problem is that the latest versions of the SD card images provided by the Red Pitaya developers set the HP0 bus width to 32. Otherwise, the code setting the HP0 bus width to 64 bits  isn't required.

I have no idea why the data isn't transferred.

blowfish2003, 2018-12-02:
Hi Pavel,
Thank you very much for the references. I have modified your adc_recorder project and was testing it when I asked the question. Using your project without modifications worked fine. Perhaps there was an issue with my modification.

closed 2018-12-06 by pavel-demin