pavel-demin/red-pitaya-notes #782: WSPR curl timeout

davidhay45, 2018-11-28:
### Description of the setup:

 * Device: Redpitaya
 * SD card image: Alpine 3.6
 * Application: WSPR
 * Other relevant information:

### Description of the problem:

curl does not upload spots and gives error in wspr log

"curl: (28) Connection timed out after 30000 milliseconds"

I have googled this and the suggestion is that dns is not resolving wsprnet.org but I can ping wsprnet.org.

This has worked before with the same setup.

The RP is wired to my router and receives its ip and dns via DHCP.

Any ideas?

pavel-demin, 2018-11-28:
The error message most probably means that the wsprnet server is overloaded and can't accept reports.

davidhay45, 2018-11-28:
I don’t think so.  I have used their website upload without issue.  The error happens every time.

From: Pavel Demin 
Sent: Wednesday, November 28, 2018 13:54
To: pavel-demin/red-pitaya-notes 
Cc: davidhay45 ; Author 
Subject: Re: [pavel-demin/red-pitaya-notes] WSPR curl timeout (#782)

The error message most probably means that the wsprnet server is overloaded and can't accept reports.

pskerrett, 2018-11-28:
Hi David,

I see the same problem as you do. I do believe it is the wsprnet site behaving slow and timing out...

One work-around you can do is to modify the curl command for a longer timeout. 

In sdr_transceiver_wspr, decode-wspr.sh file
find this line:

curl -sS -m 45 -F allmept=@$ALLMEPT -F call=$CALL -F grid=$GRID http://wsprnet.org/post > /dev/null

I changed the -m value for a longer timeout delay. (45 seconds)    YMMV..

pavel-demin, 2018-12-01:
I don't have a WSPR setup right now but I've just tried the following simple command from the latest SD card image:
```
curl http://wsprnet.org/post
```
It works without any problem on my Red Pitaya.

davidhay45, 2018-12-02:
This is weird.  When my RP is connected direct to my BT router then all works but when it is connected wired to a wifi access point then it doesn't, that is port 80 does not connect.  However when my Windows 10 pc is connected in the same way then port 80 is fine.  So the wifi extender is somehow stopping port 80 (and no doubt other ports) but only for network traffic from my RP.  Nameserver resolution and ping are fine.

This is not really the forum to ask for help on this but where is?

davidhay45, 2018-12-02:
Same problem on 3.8.  Works ok when wired to the router but not when wired to a wifi access point. So the wifi access point is blocking some ports but only for the RP machjne and does not affect my Windows 10 pc.

pskerrett, 2018-12-03:
David, something is certainly amiss on your local network somewhere.. But the problem is probably not located on your RP.  Check to make sure your getting the proper DHCP settings from your routers (Gateway, DNS, etc). But you will have to start from the network side unfortunately.

davidhay45, 2018-12-07:
This now looks like it was a hardware fault that was solid and is now just flaky.  Someone from the GitHub curl site suggested I use wireshark to capture the packets and that showed it was a corrupted return packet.  It is now working without any changes and wireshark shows the odd retransmission.

Thanks for your help.

closed 2018-12-07 by davidhay45