pavel-demin/red-pitaya-notes #783: Can't make the I2C PCA9555 inteface work with QUISK v 4.1.12

satfan52, 2018-12-06:
Hello Pavel,

I am currently using the "4 Hermes Control Bits" to switch my band-pass and low-pass filters as this works quite well with QUISK, no issues at all.

Today I decided to try instead to use I2C interface with the PCA9555 chip as I want to optimize my set-up.  Unfortunately,  I have not much success and  I wonder where is the problem.  Is the functioning of this PC9555 I2C interface application dependent ? If yes is it supported by QUISK ? Has someone tested it before ?  I am using the latest Alpine distribution and QUISK version 4.12, under  Ubuntu 16.04.

It seems to me that I have strictly followed the instructions and the schematics presented on the tutorial:
1) I made my test at Address 1 (0X21) corresponding to the following settings on the PCA9555
- pin A0 set to 3.3 V (High)
- pin A1 and A2 set to 0V (Low)  
2) The SDA and SDL pins of the PCA9555 are connected directly to the corresponding RP pins without any pullup resistors as indicated in the documentation
3) The PCA9555 chip is fed by 5V (VSS = 5V) as indicated in the Tuto (which surprised me as I expected 3.3 v)
4) I tried address 0 and 3 and the behaviour is the same 

When I reset the PCA9555 (ie cut the power and reconnect it) while the HPSDR server and QUISK are still running, then all input/output pins of the PCA9555 go to high (4.8V) and of course changing "bands" has not effects. If after that simple test, I kill both the HPSDR server and the QUISK client and then hit the reload button of the browser to restart the HPSDR server then all input/output pins of the PCA9555 go immediately to 0.56V (LOW).  Restarting QUISK has actually no effects and changing bands no effects either, all input/output pins of the PCA9555 stay in the low state "for ever"

Not sure what to try next/

Regards
Peter

pavel-demin, 2018-12-06:
Hello Peter,

It's quite possible that QUISK doesn't support the commands controlling the filters. So, I think that the very first step should be to check what commands QUISK sends to the server running on Red Pitaya.

BTW. PCA9555 should be connected to the Red Pitaya board before the application is started.

Best regards,

Pavel

satfan52, 2018-12-06:
Good Morning Pavel,

Well of course I could  put an oscilloscope on the SDR and SDA data line to see what comes out of it and/or use an I2C address scanner. But.............. I am afraid it may be some wasted time.......

Looking at this [old discussion from 2016 and 2017 on the QUISK support forum regarding the red pitaya and the topic of switching band-pass /low-pass filters](http://quisk.973856.n3.nabble.com/Low-Pass-Band-Pass-Filter-switching-td4024042.html#a4024177), it is quite clear that the red-pitaya PCA9555 I2C interface was NOT working  with QUISK at that time (in 2016/ early 2017). I checked ALL the releases notes of QUISK and didn't find any added functionality/fix addressing this very unfortunate situation. 

As a matter of fact, Jim (N2ADR) makes it crystal clear that QUISK does not manage the I2C interface of the redpitaya. [According to the Jim ](http://quisk.973856.n3.nabble.com/template/NamlServlet.jtp?macro=print_post&node=4024177) QUISK passes the band information to the red pitaya via the hpsdr protocol and "it is up to the red pitaya" to use the band information and manage accordingly all its interfaces..... 

If there is no intention to make this nice I2C interface work with the QUISK client,  neither in the QUISK client and/or in the redpitaya HPSDR server software, then I suggest to make the life of QUISK users simpler, to  indicate clearly in the redpitaya documentation of the PCA9555 I2C interface that this feature has been tested positively with PowerSDR and that TO DATE it is not currently supported by QUISK. 

Regards
Peter

pavel-demin, 2018-12-06:
I'm afraid that there is some misunderstanding somewhere.

First of all, PowerSDR mRX PS communicates with the SDR transceiver compatible with HPSDR using the HPSDR/Metis communication protocol:

https://github.com/TAPR/OpenHPSDR-SVN/raw/master/Metis/Documentation/Metis-%20How%20it%20works_V1.33.pdf
https://github.com/TAPR/OpenHPSDR-SVN/raw/master/Documentation/USB_protocol_V1.58.doc

Next, the I2C interface in the SDR transceiver compatible with HPSDR is controlled by the commands from the HPSDR/Metis communication protocol.

Since the I2C interface in the SDR transceiver compatible with HPSDR works with PowerSDR mRX PS, it means that all the commands are implemented correctly in the code running on the Red Pitaya board.

If something does work with PowerSDR mRX PS but doesn't work with Quisk, then it's obvious that the problem is in Quisk.

Jim (N2ADR) writes that "Quisk *can* send band info to the Pitaya with the Hermes Protocol". However, it's unclear if Quisk *does* send the band info by default or it should be activated somehow.

satfan52, 2018-12-06:
OK thanks for the clarifications Pavel, I will install the latest version of QUISK, check I2C signals and report results here.

pavel-demin, 2018-12-07:
I've just reread the comments in this issue and at the links from your comments and here are some new thoughts.

If Quisk controls the four Hermes Ctrl bits (pins DIO4_P-DIO7_P on the Red Pitaya board), then PCA9555 at address 0x20 should also work with Quisk. The same HPSDR/Metis commands (C0 00000000 on page 8 in [USB_protocol_V1.58.doc](https://github.com/TAPR/OpenHPSDR-SVN/raw/master/Documentation/USB_protocol_V1.58.doc)) are used to control the pins DIO4_P-DIO7_P on the Red Pitaya board and the pins P00-P06 on the PCA9555 chip at address 0x20.

PCA9555 at address 0x21 is controlled by a different HPSDR/Metis command (C0 0001001x on page 9 in [USB_protocol_V1.58.doc](https://github.com/TAPR/OpenHPSDR-SVN/raw/master/Documentation/USB_protocol_V1.58.doc)). Since this command was never mentioned in the discussion on the Quisk forum, it's possible that this command isn't implemented in Quisk.

So the right question to Jim (N2ADR) would be the following:
> Does Quisk support the HPF and LPF commands for the Alex board (command C0 0001001x on page 9 in [USB_protocol_V1.58.doc](https://github.com/TAPR/OpenHPSDR-SVN/raw/master/Documentation/USB_protocol_V1.58.doc))?

satfan52, 2018-12-07:
Thank you for this key feedback. I will ask the question of course. So logically based on this analysis  I should be able to make  Pin P00 to P06 work at address 0X20, I will test  too before I ask the question to Jim.

I started some tests yesterday evening at address 0X21 with PowerSDR as I want to see this functionality working and in action. with PowerSDR to validate my test set up. I was expecting to see the LPF and HPF pins of the PCA9555 change state as I change bands on PowerSDR.   I could not make this work though  I think I have activated the right hermes option in the PowerSDR program, at least based on a ticket opened  here in 2016. So if I can t make this work even with PowerSDR, I suspect something is wrong with my test set-up, and I need to find what because it does not make sense to test with QUISK if I can t validare my setup with PowerSDR first. Maybe my PCA9555 chip is dead or something else like that. I will do some more testing this weekend

satfan52, 2018-12-07:
OK, I was able to make the IC2 PCA9555 interface work with with PowerSDR mRX PS v 3.4.9 but only with the version of the SDR transceiver compatible with HPSDR available on the official app of red-pitaya and using the official image downloaded from STEMLAB. This works fine and like a charm, there are NO specific options to set in the set-up menu of PowerSDR, I just specified that   I used Alex during the first time installation of PowerSDR mRX PS v 3.4.9 and that is it,-

I have NOT been able to make  the IC2 PCA9555 interface work with the with the version of the HPSDR compatible SDR server available in the Alpine distribution. I need to double check that I use your very latest distribution though. A quick ssh session on the RP showed that ic2-0 is activated though.

Next tests will focuss on QUISK now that I have a working set-up with PowerSDR

pavel-demin, 2018-12-07:
Since I don't have PCA9555, I can't test if it works with the latest release.

I know that PCA9555 has been successfully used with on of the previous releases: 
https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20180523

pavel-demin, 2018-12-08:
I've just found your new post in the n2adr yahoo group. If Jim (N2ADR) doesn't have a Red Pitaya board, then I'm afraid that mentioning to him all the internal details of the I2C implementation on the Red  Pitaya board isn't the most efficient way to discuss this problem with him.

Anyway, the [Hermes-Lite2 protocol table](https://github.com/softerhardware/Hermes-Lite2/wiki/Protocol) shows that the  Hermes-Lite2 protocol implemented in Quisk isn't exactly the HPSDR/Metis protocol implemented in PowerSDR mRX PS.

Since Quisk doesn't implement all the commands of the HPSDR/Metis protocol, it explains why the I2C PCA9555 interface doesn't work with Quisk.

closed 2018-12-08 by pavel-demin

satfan52, 2018-12-31:
To the attention of QUISK users: The problem has been fixed in release 4.1.31 of QUISK.