pavel-demin/red-pitaya-notes #784: Usable bandwidth of each of the eight HPSDR receivers

bjornekelund, 2018-12-07:
CW Skimmer Server is written with the QS1R in mind, where you have a usable bandwidth of about 95% of the sample bandwidth (thus the 45.5, 91 and 182kHz bandwidths). 

Measurements show that using your software with the Red Pitaya, the usable bandwidth is closer to 89%. Beyond that sensitivity falls quite quickly. (I assume this is due to the chosen order/complexity of the DDCs. ) 

With default (14000-14091kHz etc.) settings in CW Skimmer Server, this means that performance is marginal in the first few kHz of each skimmed band. 

Knowing this, it is of course easily compensated for by shifting the receive band in CW Skimmer Server's INI file down by this amount (There is plenty of margin, already at 96kHz the bandwidth is wider than the CW portion of each ham band). 

Have you made this same observation?

I assume that, at least in theory, the new 122.88-16 model will allow for more complex DDC with a corresponding increase in usable bandwidth?

pavel-demin, 2018-12-08:
The usable bandwidth depends on the frequency response of the FIR filters. The R scripts calculating the coefficients of the FIR filters can be found in the filters sub-directory of the sdr_receiver_hpsdr project:

https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_receiver_hpsdr/filters

For the 122.88-16 board, the script will be very similar to the script from the sdr_transceiver project:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver/filters/fir_0.r

These scripts are based on the examples from the [Altera AN 455](https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/an/an455.pdf).

Pull requests with improvements to the scripts calculating the coefficients of the FIR filters are welcome.

bjornekelund, 2018-12-08:
Thanks. Very helpful. I wonder why the QS1R is different, perhaps it is just matter of FIR length. I will try to find some time to look deeper into this during xmas holidays.

pavel-demin, 2018-12-08:
Looks like the 182kHz bandwidth is possible with the new 122.88-16 board:
[image]

For comparison, here is the spectrum obtained with the current version of the application running on the 125-14 board:
[image]

bjornekelund, 2018-12-08:
Wow! That is fantastic. So I guess that means the size of the FPGA in the 125-14 is the limiting factor? Then I will not waste my time tinkering with FIR tap coefficients :-)

pavel-demin, 2019-02-24:
I've created a separate GitHub repository and a separate set of notes on the new STEMlab 122.88-16 SDR board. Here are the links:

https://github.com/pavel-demin/stemlab-sdr-notes
https://pavel-demin.github.io/stemlab-sdr-notes

The sdr_receiver_hpsdr project in this new repository has the wider 182kHz bandwidth.

closed 2019-02-24 by pavel-demin