pavel-demin/red-pitaya-notes #785: Problem with LPF and BPF swithing through Alexiares interface when QUISK controls the HPSDR server.

satfan52, 2018-12-08:
I build an Alexiares interfaces this weekend with 4*74HC595 as I wanted to compare with the I2C interface tried earlier this week to see if QUISK users are more Lucky with Alexiares than with the I2C interface.  I tested the interface with both PowerSDR and QUISK, as usual it  works perfectly with PowerSDR but there is unfortunately some problems with QUISK.

[image]

pavel-demin, 2018-12-08:
I'm afraid that I can't do anything about this problem.

Quisk implements the  [Hermes-Lite2 protocol](https://github.com/softerhardware/Hermes-Lite2/wiki/Protocol) that isn't exactly the same as the HPSDR/Metis protocol implemented in PowerSDR mRX PS and in my SDR applications. That's why the ALEX interface doesn't work with Quisk.

closed 2018-12-08 by pavel-demin

satfan52, 2018-12-31:
To the attention of QUISK  users: The problem has been fixed in release 4.1.31 of QUISK.