pavel-demin/red-pitaya-notes #786: WSPR Curl timeout (enhancement)

davidhay45, 2018-12-09:
Suggested enhancement

### Description of the setup:

 * Device: Redpitaya
 * SD card image: Alpine 3.6 but also applies to 3.8
 * Application: WSPR

### Description of the problem:
If the wsprnet.org website is slow or unavailable  then the curl upload in decode-wspr.sh script times out and simply deletes the file to be uploaded.

To fix this in the decode-wspr.sh script replace

	rm -f $ALLMEPT

with

if [ $? -eq 0 ]
then 
	echo "curl upload ok"
	rm -f $ALLMEPT
else
	echo "curl upload failed"
fi

Then the spots file is not deleted but will be uploaded at the next or later attempt.

pavel-demin, 2018-12-09:
It's how it was initially implemented. However, the wsprnet maintainers requested to remove ALL_WSPR.TXT without checking if it was successfully uploaded. Here is a link to the corresponding commit: https://github.com/pavel-demin/red-pitaya-notes/commit/830d00c008c0920b9816c8f841aabac5fa6d57e0.

closed 2018-12-09 by pavel-demin

davidhay45, 2018-12-09:
I can see that they do not want huge uploads after the wsprnet server has been down for a while, BUT it would be better to have a couple of retries built in to make it “temporary failure tolerant”.  Difficult to do in a regular running script though.

From: Pavel Demin 
Sent: Sunday, December 9, 2018 16:04
To: pavel-demin/red-pitaya-notes 
Cc: davidhay45 ; Author 
Subject: Re: [pavel-demin/red-pitaya-notes] WSPR Curl timeout (enhancement) (#786)

It's how it was initially implemented. However, the wsprnet maintainers requested to remove ALL_WSPR.TXT without checking if it was successfully uploaded. Here is a link to the corresponding commit: 830d00c.