pavel-demin/red-pitaya-notes #794: Question: axis_red_pitaya_dac data rate

Apollo3zehn, 2019-01-04:
Hi Pavel,

I have a question about your `axis_red_pitaya_dac` IP. When I simulate this IP in Vivado, I can see that the update rate of `dac_dat` is equal to the `aclk` input:

[image]

But according to the DAC [datasheet](https://www.idt.com/document/dst/dac1401d125-datasheet), it should update with twice the clock in interleaved mode. So the data should update at the same rate as the `ddr_clk` (`IQCLK`) input. 

[image]

So I would say [this](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_red_pitaya_dac_v1_0/axis_red_pitaya_dac.v#L67) line of code should be changed to `.C(ddr_clk),` to ensure the correct data rate.

What dou think? Did I overlook somethink?

Thanks
Vincent

pavel-demin, 2019-01-04:
Thanks for checking the `axis_red_pitaya_dac` module.

> What dou think? Did I overlook somethink?

Your simulated timing diagram shows the following:
 - the DAC data is updated on the falling edge of the `dac_clk` clock,
 - the rising edge of the `dac_clk` happens when the DAC data is stable.

So, your simulated timing diagram confirms that the `axis_red_pitaya_dac` module works correctly.

The internal DAC data registers (`int_dat_a_reg` and `int_dat_a_reg`) are connected to the output DDR register. The DDR register updates its output (`dac_dat`) on both the rising and falling edges of the `aclk` clock. So, `dac_dat` is update twice per the `aclk` clock period.

Apollo3zehn, 2019-01-04:
You are right, I misinterpreted my simulation. Thanks for clarification :)

closed 2019-01-04 by Apollo3zehn