pavel-demin/red-pitaya-notes #795: Request: add channel/antenna parameter to write-c2-files name

waterwin, 2019-01-16:
In the multiband decoders the c program <write-c2-files> names its output as follows:
ft8_7_50314500_190115_1708.c2

I am using the multiband decoders to compare at almost the same time two antenna's on the inputs #1 and #2.

For my analysis it would help a lot if the files are created with the used antenna in their filename.

F.i. ft8_7_50314500**_1_**190115_1708.c2

The line 159 in the program would need to be changed to add %c or something for the channel parameter. 
sprintf(name, "ft8_%d_%d_%s.c2", i, (uint32_t)dialfreq, date);

It would also need a very small adjustment in the decode-ft8.sh to pick up the files with the changed name.

Currently I do the analysis based on the timeframe. I have defined the 1st and 3rd quarter of an hour for antenna 1 and the 2nd and 4th quarter for antenna 2. That is an indirect workaround and will not work of the .cfg files contain a mix of bands listening to input/antenna 1 and 2.

pavel-demin, 2019-01-26:
I'm not convinced that the modifications that you suggest should be added to my repository.

If you know exactly what lines should be modified, then why you don't modify them in your copy of the code?

waterwin, 2019-01-26:
Dear Pavel, thank you for your reaction. Some thoughts on this from my side:
1. It is a C program and I have no experience in modifying C code. 2. The data is already available and is now discarded.3. I think others besides myself will also have value from it and therefore it is a addition to the original repository.4. It will make  both tools (FT8 and WSPR) even better for their task in comparing the results of different antenna's/inputs.
I do this now based on a time frame-differentiation in your scripts. Running antenna 1 for 1st and 3rd quarter and antenna 2 in quarter 2+4. But that is indirect and prone to errors on the boundaries of the periods.
Kind regards, please let me know how I can be of help.
Erwinde PE3ES

pavel-demin, 2019-01-27:
With [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/f7d5cfaae8b4b2a25560d94355784ba00fce8c2e), I've added the channel numbers to the c2 file names.

Here is a link to a pre-release version of the SD card image containing the new code:

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AABJiwe_GkAtx0SbzNsyLOyOa/sdr/red-pitaya-alpine-3.8-armhf-20190127.zip?dl=1

Please test it and let me know how it works.

waterwin, 2019-01-27:
Thanks a lot Pavel, I will test the result the coming days.
ErwinPE3ES

 
 

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AABJiwe_GkAtx0SbzNsyLOyOa/sdr/red-pitaya-alpine-3.8-armhf-20190127.zip?dl=1

waterwin, 2019-01-27:
Brilliant so far. Will work from here to get the data split (see enclosed screenshot for the 3 files from chan #2.
Thanks againErwin

 
 

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AABJiwe_GkAtx0SbzNsyLOyOa/sdr/red-pitaya-alpine-3.8-armhf-20190127.zip?dl=1

pavel-demin, 2019-01-27:
Thanks for testing this new version.

GitHub doesn't accept attachments sent via e-mail. It's only possible to add screenshots via the web interface.

waterwin, 2019-01-31:
[image]

Using the new setup pointed me to a hardware issue. It appears that my channel 2 is dead. When I tested it with the standard Red Pitaya setup Oscilloscope it remained dead.

Conclusion: the change in the program delivers c2 files with a filename showing the input channel used.

Now I have to find a solution for my broken channel.

pavel-demin, 2019-02-15:
There is now a release with the updated multiband FT8 and WSPR transceivers that output c2 files with the  channel numbers in the file names:
https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20190215

closed 2019-02-15 by pavel-demin

waterwin, 2019-02-15:
THANK YOU

Highly appreciated

Erwin/PE3ES