pavel-demin/red-pitaya-notes #802: Can't boot Embedded Image

LeLyfa, 2019-02-13:
### Description of the setup:

 * Device: RedPitaya w/ 14-Bit ADC
 * SD card image: Downloaded from https://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-wspr/
 * Other relevant information:

I can't boot, here' the error code from the serial console.

`
mmc0: Problem switching card into high-speed mode!
mmc0: new SDHC card at address 0001
mmcblk0: mmc0:0001 SD16G 3.73 GiB
 mmcblk0:
Alpine Init 3.3.0-r1
  Loading boot drivers: ok.
  Mounting boot media: random: fast init done
ok.
  Loading user settings from /media/mmcblk0/red-pitaya.apkovl.tar.gz: ok.
  Installing packages to root filesystem: WARNING: Ignoring /media/mmcblk0p1/apks/armhf/APKINDEX.tar.gz: No such file or directory
fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/main/armhf/APKINDEX.tar.gz
ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.8/main: temporary error (try again later)
WARNING: Ignoring APKINDEX.adfa7ceb.tar.gz: No such file or directory
fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/community/armhf/APKINDEX.tar.gz
ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.8/community: temporary error (try again later)
WARNING: Ignoring APKINDEX.efaa1f73.tar.gz: No such file or directory
OK: 0 MiB in 0 packages
ok.
grep: /sysroot/etc/inittab: No such file or directory
/sbin/init not found in new root. Launching emergency recovery shell
Type exit to continue boot.
sh: can't access tty; job control turned off
`

pavel-demin, 2019-02-13:
I've just downloaded the red-pitaya-alpine-3.8-armhf-20181218.zip file and tested it with my Red Pitaya board. It works without any problem. So, I'd say that the file is OK.

The very first line (`mmc0: Problem switching card into high-speed mode!`) in your logs looks strange. Could it be an indication of a problem with the SD card? 

I'd suggest to check the following two points:
 - the SD card doesn't have any hardware problem
 - all the files and directories from the red-pitaya-alpine-3.8-armhf-20181218.zip file are copied to the SD card

Here is how my logs look like:

```
mmc0: new high speed SDHC card at address 1234
mmcblk0: mmc0:1234 SA16G 14.4 GiB
 mmcblk0: p1
Alpine Init 3.3.0-r1
 * Loading boot drivers: ok.
 * Mounting boot media: usb 1-1: new high-speed USB device number 2 using ci_hdrc
random: fast init done
ok.
 * Loading user settings from /media/mmcblk0p1/red-pitaya.apkovl.tar.gz:   0%                                                                            ok.
 * Installing packages to root filesystem: fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/main/armhf/APKINDEX.tar.gz
ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.8/main: temporary error (try again later)
fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/community/armhf/APKINDEX.tar.gz
ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.8/community: temporary error (try again later)
(1/79) Installing musl (1.1.19-r10)
(2/79) Installing busybox (1.28.4-r2)
```

The ERROR messages are OK since all the necessary files are present on the SD card.

pavel-demin, 2019-02-13:
Here is a list of all the files and directories that should be present on the SD card:
```
apks
apps
boot.bin
cache
devicetree.dtb
modloop
red-pitaya.apkovl.tar.gz
uEnv.txt
uImage
uInitrd
wifi
```

closed 2019-02-18 by pavel-demin