pavel-demin/red-pitaya-notes #803: Unable to boot Alpine 3.8 OS

alandong93, 2019-02-15:
### Description of the setup:

 * Device: STEMLab 125-14
 * SD card image: red-pitaya-alpine-3.8-armhf-20181218
 * Application: High-speed direct sampling voltage measurements (want to interface with GRC eventually)
 * Other relevant information: I am attempting to write the SD card image files to the SD card that originally shipped with the RP from a Windows 10 laptop using [ImgBurn](http://www.imgburn.com/) and [Win32DiskImager](https://sourceforge.net/projects/win32diskimager/).

### Description of the problem:

After formatting the SD card and writing an image file containing the files from red-pitaya-alpine-3.8-armhf-20181218.zip to the SD card, I insert the card into the RP and power it up. The green power LED lights up and the orange LED (LED 8) lights up partially, but no other LEDs light up.

### Steps to reproduce the problem:

 1. Download [red-pitaya-alpine-3.8-armhf-20181218.zip](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20181218/red-pitaya-alpine-3.8-armhf-20181218.zip).
 2. Unzip the files (with 7-Zip).
 3. Run ImgBurn as administrator. Add the unzipped files and write a .img file. It takes a few seconds. ImgBurn also produces a .mds file.
 4. Insert the SD card into the laptop. Windows 10 prompts you to format it. Capacity says "3.72 GB". Select "FAT32 (Default)". Select "Default allocation size". Uncheck "Quick Format". Click "Start". It takes several minutes.
 5. Run Win32DiskImager as administrator. Select the .img file. Select the SD card drive. Click "Write". It takes a few seconds.
 6. Insert the SD card into the RP. Power it on. The green power LED lights up and the orange LED (LED 8) lights up partially, but no other LEDs light up.

Any help would be appreciated so, so much! Thank you for all the work you do, Pavel!

-Alan

pavel-demin, 2019-02-15:
Hello Alan,

Since you mention GRC, I suppose that you're planning to use the [SDR transceiver](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver) application and you're following the [Getting started with GNU Radio](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/#getting-started-with-gnu-radio) instructions.

The steps 1, 2 and 6 from your list are correct. The steps 3, 4 and 5 aren't correct. 

Copying the content of the SD card image zip file to an SD card doesn't require  ImgBurn nor Win32DiskImager.

After unzipping red-pitaya-alpine-3.8-armhf-20181218.zip, the files can be copied using the copy and paste functionality of the Windows 10 File Explorer:

- Open folder with the files and folders from red-pitaya-alpine-3.8-all armhf-20181218.zip
- Select all the files and folders from red-pitaya-alpine-3.8-armhf-20181218.zip
- Press Ctrl+C
- Open SD card
- Press Ctrl+V

The files from red-pitaya-alpine-3.8-armhf-20181218.zip should be copied to an empty FAT formatted SD card.

Best regards,

Pavel

alandong93, 2019-02-15:
Hi Pavel,

Thank you for your response. I just tried this with an empty FAT-formatted SD card and now the behavior has changed. The blue LED is on, but LED 0 (orange) is blinking and I don't believe the OS is booting. I also tried the same instructions but with FAT32 instead of FAT and observed the same behavior.

-Alan

pavel-demin, 2019-02-16:
> LED 0 (orange) is blinking

It's already a good sign. It means that the Zynq chip was able to read the boot.bin file from the SD card.

The boot.bin file loads the LED blinker configuration to the FPGA and starts the OS booting process.

> I don't believe the OS is booting

I don't see why it wouldn't boot. Have you tried to connect to the web interface or to the USB/serial console?

I'd suggest to try the following next steps:
- connect to the on-board web interface
- try to run one of the applications via the web interface

closed 2019-02-18 by pavel-demin

alandong93, 2019-02-21:
Hi Pavel,

I do not see the red LED (CPU heartbeat), which is why I think it's not booted correctly. I cannot connect to the web interface when plugged into the RP with Ethernet. My laptop Ethernet interface knows that something is connected because when I power off the RP, `ipconfig /all` shows `Media disconnected`, but I cannot see the RP Ethernet interface's MAC address with `arp -a` in Powershell.

-Alan

pavel-demin, 2019-02-21:
> I do not see the red LED (CPU heartbeat), which is why I think it's not booted correctly.

The CPU heartbeat isn't enabled in my configuration. So, it's OK that the red LED isn't blinking.

> My laptop Ethernet interface knows that something is connected because when I power off the RP, ipconfig /all shows Media disconnected.

Do I understand correctly that you connecting the Red Pitaya Ethernet port to your laptop Ethernet port with a single cable?

If it's so and if there is no DHCP server running on your laptop, then in this setup the IP address of the Red Pitaya board is 192.168.1.100. It's explained in my notes about Alpine Linux:
http://pavel-demin.github.io/red-pitaya-notes/alpine/#network-configuration

The IP address of the Ethernet port of your laptop should be manually set to an address in the 192.168.1.x network. For example, 192.168.1.101.

I'd also suggest to disconnect the laptop from the Wi-Fi network to make sure that there is no collisions of the IP addresses in the wired and the wireless networks.

In my opinion, it would be easier to connect both the laptop and the Red Pitaya board to a router. This way, the router would assign IP addresses to the laptop and to the Red Pitaya board via the DHCP protocol.

alandong93, 2019-02-21:
Hi Pavel,

Thank you for your help. I think what was wrong was my Ethernet IP and subnet mask, and that the OS was booting correctly. I figured out a different solution, so I won't try what you suggested. I am now using the default RP Ubuntu OS. I downloaded your sdr_transceiver application from the bazaar and it's working for me beautifully. I can't thank you enough for all the work you do to support the RP community! Thank so much!

Best,
Alan