pavel-demin/red-pitaya-notes #808: Hopefully interesting hint: SDR ADALM Pluto

df8oe, 2019-02-25:
Hi Pavel,

not an issue at all - but something which might be interesting. Adalm Pluto is a SDR TRX which works from 70MHz...6GHz in full duplex. It uses AD9363 transceiver chip and same Zync FPGA-SOC as RP - so I think most of your work is adaptable in a not very difficult way. The hardware costs are ~140,-- Euros and it starts at the frequency RP ends so it would be a very nice addition. Adding your multiwspr and  HPSDR-TRX would cover very interesting bands (I am talking about new geostationair amateur radio satellite Es'Hail) and would give Adalm Pluto full compatibility to HPSDR software. A network device over USB is already present... I am not sure if you are interested to support this device, too. If not: Maybe here are some reading who wants to start with such a project?

pavel-demin, 2019-02-25:
I think that this kind of requests should be addressed to the developers who are already working with ADALM-PLUTO:
https://github.com/unixpunk/PlutoWeb

I know about Adalm Pluto. At some point, I was interested to buy it for under 100 euros but it wasn't easily available. Now, it's easily available but it's more expensive.

Since Adalm Pluto was designed as a very basic learning module, compared to the Red Pitaya boards, Adalm Pluto has some limitations:
 - no microSD card slot
 - very limited on-board flash memory (32 MB)
 - very limited number of GPIO pins
 - no on-board Ethernet ports

At the moment, I'm not very interested in developing anything for such a limited device. I'd rather use the applications developed by others.

closed 2019-02-25 by pavel-demin

df8oe, 2019-02-25:
Hi Pavel,
many thanks for coming back to my posting. Of course I understand your argumentation. Let me tell you that I enjoy your really nice work on RP very much - I love it! Good luck to you and I stay tuned for everything what will come in the future from you :)

satfan52, 2019-02-25:
I own a Pluto and I confirm that to date there are quite a few SDR transceiver applications that work with it already: SDR-Angel, SDR-Console 3 and I understand that development is ongoing for linhpsdr and  QUISK (via the soapySDR module). So not too sure what a port of the hpsdr compatible transceiver application could bring to the party. However what could be a true "killing" is a port  of the red-pitaya VNA application to the Pluto (if that is technically feasible)

pavel-demin, 2019-02-25:
> However what could be a true "killing" is a port of the red-pitaya VNA application to the Pluto (if that is technically feasible)

I think that this kind of requests should be addressed to the developers of the ADI IIO Oscilloscope and the default FPGA configuration for ADALM-PLUTO:

https://github.com/analogdevicesinc/iio-oscilloscope
https://github.com/analogdevicesinc/hdl/tree/master/projects/pluto

Maybe they would be interested in adding the VNA functionality to the ADI IIO Oscilloscope.

satfan52, 2019-03-08:
Actually, the Scopy app under development will be supported by the Pluto and will  offer a network analyser app. https://github.com/analogdevicesinc/iio-oscilloscope/issues/143#issuecomment-470993465