pavel-demin/red-pitaya-notes #814: Documentation

sixtemesseven, 2019-03-18:
Hi

I'm testing your mcpha software and I get it to work.
However, I find it somewhat frustrating that there is very little documentation on the features of mcpha.

First of, I do understand that this is completely free software and so on. But i find it frustrating that such a pice of software which a lot of work has gone into falls flat because of documentation...

A couple of things:

1. How does:
- Baseline subtraction mode
- decimator factor
Work in the GUI?

2. Is there a possibility to switch between HV and LV on the red pitaya while running mcpha?

3. What are the signal input requirements? Does it expect the output from a charge sensitive pre amplifier or does it expect to get signals from a pulse sharpener? In other words how does it deal with stacked pulses? If so what filter type does it [implement]([url](url
[image]
)) (triangle I asume?) and how do I set the dsp parameters?

I'm feeding in a signal from a preamplifier but I only get somthing which resembles a pink noise curve. If I attach a signal generator and feed in clean pulses I get very nice spikes though. So either the PHR of my scintillator / pmt is terrible or I do not understand your software.

pavel-demin, 2019-03-18:
Thanks for your interest in the MCPHA application and for your feedback.

Looking at the screenshot, I'd say that the pulse amplitude is too low.

Could you please post a screenshot of the oscilloscope tab? It'll give me an idea of the input pulse shape.

Concerning the documentation. The development of my projects for Red Pitaya depends on the feedback and contributions of the users. So far, nobody complained about the MCPHA documentation and I thought that the application is simple enough to be used with the information already contained in my notes.

> How does baseline subtraction mode works

In the manual mode, it just subtracts the value entered via the 'baseline level' field from the peak value.

In the auto mode, the base line level is defined as the minimum value just before the rising edge.

> How does decimator factor works

The 'decimation factor' field controls the decimation factor of the CIC filter. It's used to adjust the sampling rate and thus the width of the pulse at the input of the pulse height analyzer module. 

BTW. The oscilloscope tab can be used to check the shape of the pulse at the input of the pulse height analyzer module. 

> 2. Is there a possibility to switch between HV and LV on the red pitaya while running mcpha?
> 3. What are the signal input requirements?

The signal input requirements are similar to the requirements of other similar devices.

The MCPHA application works well with Gaussian-shaped pulses. It also works with exponentially rising and decaying pulses if the rise time is about a few tens of nanoseconds and the decay time is about a few tens of microseconds.

The position of the HV/LV jumper depends on the amplitude range of the input signal. Ideally, the amplitude range of the signal should closely match the hardware input range.

> Does it expect the output from a charge sensitive pre amplifier or does it expect to get signals from a pulse sharpener? In other words how does it deal with stacked pulses?

It expects a signal from a pulse-shaping amplifier.

In its current state, the MPCHA application doesn't have any pulse shaping module. However, it should be relatively easy to add a simple pulse clipping module that would delay, scale and subtract the delayed signal from the undelayed signal as it's for example described in  [Digital signal processing for BGO detectors](http://dx.doi.org/10.1016/0168-9002(93)91105-V).

Normally, it's not designed to work with stacked pulses. However, you can try to set the baseline subtraction mode to 'auto' and check if the performance is acceptable for your project.

sixtemesseven, 2019-03-19:
Thank you for your long replay and sorry for my delayed answer. I managed to kill my preamp in between.

I did not realize that LV and HV are hardware options since in the oscilloscope module (the preinstalled one) there is a HV LV switch so I assumed there was a range relay...

I attached some of the screenshots. Most of the peaks are pretty low but there are some which reach 10V (Now on the HV setting). I will add another amplifier and get it into the 20V range.

[image]
[image]

I had little luck with the baseline set to auto so far. When I probe with my real oscilloscope I see a "noise floor" of about 0.4. With no baseline and a decimator factor of 20 I get the spectrum attached.

[image]

pavel-demin, 2019-03-19:
Thanks for the new screenshot.

I've reread my previous comment and found an error in my description of the exponentially decaying pulses. The decay time of the pulses that I tested was around one microsecond.

Here are the parameters that worked for me:
- rise time 50 ns
- decay time 1 us
- decimation factor 4

The pulses on the new screenshots have a rise time of about 20 microseconds and a decay time of about 500 microseconds.

I'd try to set the decimation factor to 2048.

If it doesn't help, then the only solution that works with the current version of the MCPHA application is to add a pulse-shaping amplifier to your setup.

> I did not realize that LV and HV are hardware options since in the oscilloscope module (the preinstalled one) there is a HV LV switch so I assumed there was a range relay...

There is no range relay. Each input has a couple of jumpers that can be used to select the input range. The description of the jumpers can be found in the Red Pitaya documentation:

https://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/fastIO.html

sixtemesseven, 2019-03-19:
2048 works a lot better in particular on the lower end.

What is your formula for the decimator factor calculation?

I changed the preamp. resistor and now I have about 10us rise ad 100us decay time for a 2V pulse.

pavel-demin, 2019-03-19:
> What is your formula for the decimator factor calculation?

Unfortunately, I don't have any formula for the decimation factor. I think that it should be set to a value that maximizes the spectrum resolution.

For the factor 2048, I've just scaled the factor 4 that worked for the 1 us pulses by approximately the ratio of the 500 us and 1 us decay times.

So, for the 100 us decay time, I'd try to set the decimation factor to a value of around 400.

sixtemesseven, 2019-03-19:
I have added a CR-RC filter and got the fall time down to about 10us. Still experimenting on the settings.

However, I found another curious thing with the scaling of the oscilloscope: It seems to autoscale but then the data goes out of range...

[image]

sixtemesseven, 2019-03-19:
One other thing I just saw:

Why is the signal (I asume) clipping althought there should 16300 bins? Decimator factor set to 25.

[image]

pavel-demin, 2019-03-19:
> It seems to autoscale but then the data goes out of range...

All the plots can be manually zoomed with the left mouse button and unzoomed with the right mouse button. There is no autoscale in the oscilloscope tab.

> Why is the signal (I asume) clipping althought there should 16300 bins? Decimator factor set to 25.

I think that the amplitude of the signal after the CIC filter depends on the decimation factor. Maybe a multiplier should be added to the FPGA configuration to compensate for this kind of effects.

sixtemesseven, 2019-03-21:
> In its current state, the MPCHA application doesn't have any pulse shaping module. However, it should be relatively easy to add a simple pulse shaping module that would delay, scale and subtract the delayed signal from the undelayed signal as it's for example described in Digital signal processing for BGO detectors and in Digital gamma-ray spectroscopy based on FPGA technology.

That sounds like a good idea. I have built a analog delay line based shaper + a gaussian shaper and it does work but it seems unreasonable to do this in the analog domain while having that much dsp power.

I tried looking at the block diagram of your mcpha of the fpga which I assume is the: block_design.tlc located in: \red-pitaya-notes-master\projects\mcpha

Howeber if I try to run it in vivado 2018.2 via source block_design.tlc as described in your examples I get the error:

> source block_design.tcl
# cell xilinx.com:ip:clk_wiz pll_0 {
#   PRIMITIVE PLL
#   PRIM_IN_FREQ.VALUE_SRC USER
#   PRIM_IN_FREQ 125.0
#   PRIM_SOURCE Differential_clock_capable_pin
#   CLKOUT1_USED true
#   CLKOUT1_REQUESTED_OUT_FREQ 125.0
#   CLKOUT2_USED true
#   CLKOUT2_REQUESTED_OUT_FREQ 250.0
#   CLKOUT2_REQUESTED_PHASE -90.0
#   USE_RESET false
# } {
#   clk_in1_p adc_clk_p_i
#   clk_in1_n adc_clk_n_i
# }
invalid command name "cell"
    while executing
"cell xilinx.com:ip:clk_wiz pll_0 {
  PRIMITIVE PLL
  PRIM_IN_FREQ.VALUE_SRC USER
  PRIM_IN_FREQ 125.0
  PRIM_SOURCE Differential_clock_capable_pin
  C..."
    (file "block_design.tcl" line 2)

pavel-demin, 2019-03-21:
I'd prefer to keep this issue open until I update the notes about the MCPHA application.

pavel-demin, 2019-03-21:
> Howeber if I try to run it in vivado 2018.2 via source block_design.tlc as described in your examples I get the error

What OS is installed on your development machine?

If it's Linux, then the commands to build adc_recorder_trigger.bit are
```
cd red-pitaya-notes
make NAME=mcpha xpr
```

If it's Windows, then open Vivado, and in the Tcl console, paste the following lines:
```
cd c:/#your-path-to-the-code#/red-pitaya-notes-master
source helpers/build-cores.tcl
set argv [list mcpha xc7z010clg400-1]
source scripts/project.tcl
```

sixtemesseven, 2019-05-22:
Hi Pavel

Thank you for your help so far!
I have added a FIR filter just before the CIC stage's of one channel with a triangular filter coefficents as described here:
http://web.engr.oregonstate.edu/~tavakola/NE-RHP537_files/L2_Filters.pdf

I have a bitstream file from Vivado (Windows). Can I save the bitstream file to the red-pitaya-notes on my Debian machine and compile an sd image from there directly? Kind of like you describe it in the led_blink project. In which folder would I have to put the generated temp/mcpha.runs/system_wraper.bit file?

My red_pitaya_notes_master directory does not contain a /dev directory like described in the led_blink project...(?)

Thank you once more

pavel-demin, 2019-05-22:
I'd suggest to use the latest version of the [SD card image based on Alpine Linux](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20190215/red-pitaya-alpine-3.9-armv7-20190215.zip) and copy your .bit file to `apps/mcpha/mcpha.bit`.

In the LED blinker notes, `/dev/mmcblk0` is the path to the SD card device. If you don't have it then just skip this command and use the next one to build a SD card image.

sixtemesseven, 2019-05-26:
Trying to run alpine:

Step: _Copy the content of the SD card image zip file to an SD card_

I tried to unzipp and copy which did not work. Yellow LED blinks, green and blue are solid. LAN is blinking ans showing packages. However, it does not show up on my local net or on on a seperate LAN adapter on port 192.168.1.100

I read on the rasp. pi page that one has to unzipp the tar.gz file into the sd card as well. So I unziped both which gave some error in windows about permission and paths. Will try on Ubuntu again.

However, this step is not clear to me...

pavel-demin, 2019-05-26:
>  Yellow LED blinks, green and blue are solid. LAN is blinking ans showing packages.

This is OK.

> However, it does not show up on my local net or on on a seperate LAN adapter on port 192.168.1.100

More details about your network setup would certainly help to understand this problem.

I'd suggest to check the boot process and the IP address of the Red Pitaya board using the USB/serial console:

https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html

> I read on the rasp. pi page that one has to unzipp the tar.gz file into the sd card as well.

This is wrong. The Raspberry Pi instructions can't be used with this SD card image.

pavel-demin, 2019-05-26:
I've just updated the notes about the MCPHA application with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/a9294f0347f84b05731b0378c11158aae63a1049). So, this issue can be closed.

sixtemesseven, 2019-05-26:
Thank you for your kind help!

closed 2019-05-26 by sixtemesseven