pavel-demin/red-pitaya-notes #818: idiots guide

jeffforber, 2019-03-23:
### Description of the setup     Wndows 10 ....no linux pc 

 * Device: redpitaya 125-14 
 * SD card image: red-pitaya-alpine-3.9-armv7-20190215.zip
 * Application:
 * Other relevant information:   i have the hdsdr and power sdr working ok ....iam 75 and  i have no programing skills  ok with a raspberry -pi copy and paste ........ unpacked the file with 7zip   ,,when i tried to flash program it says no .img  so i just copied and pasted the file to the sd card 
  i put my call sign and grid location and copied the start file too the first dir 
### Description of the problem:i followed your instuctions but the green led is only on .... ...i get rp is ready for wspr message but no other leds light up .....  i hope this is enough info .........  i want to use wspr and ft8 rx.... i would really like a step by step idiots guide please 
 regards jeff  TA4/G8SCU

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-03-23:
> when i tried to flash program it says no .img so i just copied and pasted the file to the sd card

All the files from and directories should be copied from the .zip file to a SD card.

> i put my call sign and grid location

What text editor are you using to edit decode-wspr.sh?

jeffforber, 2019-03-23:
Windows note pad

jeffforber, 2019-03-23:
Do you need to extract the files??

On 23 Mar 2019 20:47, jeff forber <jeff.forber@hotmail.co.uk> wrote:

pavel-demin, 2019-03-23:
> Do you need to extract the files?

Yes, you need to extract all the files and directories from the .zip file to a SD card.

pavel-demin, 2019-03-23:
I think that the editor that you used to edit decode-wspr.sh replaced the Unix end-of-line characters with the Windows end-of-line characters.

I'd suggest to use an editor that doesn't change the end-of-line characters. For example, Notepad++:

https://notepad-plus-plus.org

Since it starts to be a recurrent problem, I've just tried to add a workaround of this problem to my scripts. Here is a link to a .zip file that contains this workaround: 

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADmGGMgiQrRUOAM1LYE41Mna/red-pitaya-alpine-3.9-armv7-20190323.zip?dl=1

Could you try to download and install this new version?

jeffforber, 2019-03-23:
Ok thank you for getting back so quickly i will try this .advice

https://notepad-plus-plus.org

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADmGGMgiQrRUOAM1LYE41Mna/red-pitaya-alpine-3.9-armv7-20190323.zip?dl=1

jeffforber, 2019-03-23:
Hello Pavel.....i have tried that latest version. Plus i used a different sd card its not working just a solid green led.

https://notepad-plus-plus.org

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADmGGMgiQrRUOAM1LYE41Mna/red-pitaya-alpine-3.9-armv7-20190323.zip?dl=1

jeffforber, 2019-03-23:
I will try the note pad ++

https://notepad-plus-plus.org

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADmGGMgiQrRUOAM1LYE41Mna/red-pitaya-alpine-3.9-armv7-20190323.zip?dl=1

pavel-demin, 2019-03-23:
> Plus i used a different sd card its not working just a solid green led.

I don't see any problem with the solid green LED. It was once requested to disable the blinking LEDs in my applications to reduce the noise level.

pavel-demin, 2019-03-23:
I see that your multiband spots start to appear in the wsprnet.org fatabase. So, the WSPR application works.

jeffforber, 2019-03-23:
Yes i see now its working ...fantastic.

jeffforber, 2019-03-23:
Tomorrow i will try Ft8 ...thank you very much ....rrgards jeff

closed 2019-03-24 by pavel-demin