pavel-demin/red-pitaya-notes #820: SSH

jeffforber, 2019-03-27:
### Description of the setup:

 * red-pitaya-alpine-3.9-armv7-20190323.zip
 * Application:idiots gude
 * Other irrelevant information:   hello Pavel another request please .....i can not believe how well this little box of tricks works, i wish i had a fraction of the brain you have .....you are a genius 

### Description of the problem:                                           
 as a complete newbie/idiot...how to change bands using SSH.....???? 

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-03-27:
Here are the instructions:
- connect to Red Pitaya following the instructions from the Red Pitaya documentation:
https://redpitaya.readthedocs.io/en/latest/developerGuide/os/ssh/ssh.html
- make SD card writable
```
rw
```
- edit write-c2-files.cfg
```
nano apps/sdr_transceiver_wspr/write-c2-files.cfg
```
- make SD card read-only
```
ro
```

jeffforber, 2019-03-27:
Thank you very much for the quick response....regards jeff

https://redpitaya.readthedocs.io/en/latest/developerGuide/os/ssh/ssh.html

closed 2019-03-30 by pavel-demin