pavel-demin/red-pitaya-notes #822: Red Pitaya TDC crashes

mikepompi, 2019-04-08:
Hello,

I've been working on implementing a TDC (on the FPGA side) which sends data to a web application. The problem is, that when I run both the bit file and the web application the Red Pitaya freezes and crashes (the website and the terminal line are frozen). Separately they both run as intended.

I've been following the official [guide](https://redpitaya.readthedocs.io/en/latest/developerGuide/fpga.html#) to create a basic template (the project Classic). Then I created a Vivado project and added a TDC IP (custom) and an AXI GPIO IP. I then generated the bitstream, exported hardware and launched the SDK. There I created the device tree and FSBL. I moved all of the files to a template app on Red Pitaya, modified the fpga.conf file and when I ran the app online Red Pitaya stopped working.

pavel-demin, 2019-04-08:
How is it related to my notes and projects?

mikepompi, 2019-04-08:
It is not directly. I couldn't get access to Red Pitaya forum, to post the issue there, so I thought that someone here might see the solution to my problem, of how to combine the web and the fpga part.

pavel-demin, 2019-04-08:
OK. I see.

The system freezes if a program can't communicate with the FPGA. It could happen if the main clock used my the FPGA configuration isn't running or if there is a problem with the AXI bus.

I'd start by checking the clock.

closed 2019-04-11 by pavel-demin