pavel-demin/red-pitaya-notes #824: Smith Chart

skedward74, 2019-04-11:
### Description of the setup:

 * Device: VNA
 * SD card image: Red Pitaya Alpine 3.9 Arm V7 20190215
 * Application: VNA
 * Other relevant information:

### Description of the problem:
When running the VNA application the smith chart produces incorrect results. Results as per attached file. This was with a 50 Ohm load.

[image]

pavel-demin, 2019-04-11:
Please provide more details about your setup:
 - Red Pitaya board model (125-14 or 125-10).
 - How the Red Pitaya board is connected to DUT?
 - How the calibration was done? 
 - A cfg file corresponding to this measurement.

skedward74, 2019-04-12:
- Model 125-14
- The smith chart above is from the supplied 50Ohm load connected directly to the DUT port of the VNA
- Open, short, load (50 Ohm)
- cfg file attached

Thanks

[vna.zip](https://github.com/pavel-demin/red-pitaya-notes/files/3074438/vna.zip)

pavel-demin, 2019-04-12:
Are you using the VNA module sold by the Red Pitaya company? If so, then according to its specifications at [this link](https://www.redpitaya.com/f172/vector-network-analyzer-module), this module only works starting from 500 kHz.

I've checked the vna.ini file. The load and DUT data are quite noisy for the frequencies lower than 2 MHz.

The only solution that I can propose is to set the start frequency to 2 MHz. Maybe 1 MHz will also work.

pavel-demin, 2019-04-12:
I've just had another look at the vna.ini file.

I'd say that the levels of the signals in the open and short data are very low. Less than 0.001. I'd expect them to be much higher. Somewhere between 0.1 and 1.0.

I'd suggest to check the following:
 - connections between the VNA module and the Red Pitaya board
 - input jumpers on the Red Pitaya board (they should be in the LV position)

skedward74, 2019-04-12:
Yes its the VNA sold by red Pitaya, and yes, my fault sorry. Had my start frequency set too low. Recalibrating between 2MHz and 60Mhz gives more meaningful results. Still quite noisy below that but I guess that's just limitations of the device (or maybe my setup).
Thanks

pavel-demin, 2019-04-12:
> Still quite noisy below that but I guess that's just limitations of the device (or maybe my setup).

The noise is due to very low level of the input signals. In the vna.ini file they are 100 times lower than they should be.

Please check the points from my previous comment.

skedward74, 2019-04-12:
Problem solved:
One of the SMA cables supplied with the VNA has no centre pins in the SMA connectors (my old eyes didnt spot this earlier). Replaced cable and now unit works down to 500KHz.
Thanks for your help.

closed 2019-04-12 by pavel-demin