pavel-demin/red-pitaya-notes #828: I get stuck in a loop k

ssooffiiaannee, 2019-04-16:
### Description of the setup:

 * Device: Red Pitaya
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:
In the led blinker tutorial, I execute de command make NAME=led_blinker all, and I get stuck 
in a loop that restarts undifinitely, I'm working on it for the past days, and I'm still stuck on it.

### Steps to reproduce the problem:
after installing vivado: 
 1.source /opt/Xilinx/Vivado/2018.3/settings64.sh
 2.git clone https://github.com/pavel-demin/red-pitaya-notes
cd red-pitaya-notes
 3.make NAME=mcpha bit

pavel-demin, 2019-04-16:
What operation is repeated? Why do you think that Makefile execution is repeating itself indefinitely? Could you provide some log or screenshot illustrating the problem?

ssooffiiaannee, 2019-04-16:
I installed vivado
In Red Pitaya notes, In the led blinker, I executed the following commands : 
[image]
 
Then I get the following result that is repeating  (it's not the whole result I put in this image, since I cannot put everything in on single image )
[image]

ssooffiiaannee, 2019-04-16:
Could you tell me what is the next step in Build SD card image zip file, in mcpha project?

[image]

pavel-demin, 2019-04-16:
Thanks for the screenshot.

When you run the make command for the first time, it's normal that it runs the `core.tcl` script multiple times. The `core.tcl` script is used to build the IP cores from the sources that can be found in the [cores](https://github.com/pavel-demin/stemlab-sdr-notes/tree/master/cores) directory. Since there are 49 IP cores in the [cores](https://github.com/pavel-demin/stemlab-sdr-notes/tree/master/cores) directory, the `core.tcl` script is executed 49 times.

pavel-demin, 2019-04-16:
The `build-all.sh` script does the following:
 - builds the IP cores
 - builds led_blinker.bit
 - builds Linux
 - builds U-Boot
 - builds all the projects that I normally put on my SD card images
 - builds a SD card image based on Alpine Linux

ssooffiiaannee, 2019-04-16:
Thank you for your reply
My first question is: Do I have to write my SD card after this command or is it automatically done ?
My second one : Do I have to insert the SD card while executing this command?

pavel-demin, 2019-04-16:
The `build-all.sh` creates a zip file. The content of the zip file should be copied manually to an SD card. It's the same zip file as linked from the 'Getting started' instructions.

ssooffiiaannee, 2019-04-16:
I actually have 2 partitions on my SD card, to which partition shall I copy it?

In the led blinker tutorial, what does this command do? because I don't have mmcbl0 in the dev directory, even after inserting my SD card

[image]

pavel-demin, 2019-04-16:
I normally use a SD card with a single FAT formatted partition. It's how most of the SD cards are formatted out of the box.

The command does exactly that is written: builds a bootable SD card. Where `/dev/mmcblk0` is the path to the SD card. If you don't have it then skip this command and try the next one that builds a SD card image that can be later copied to a SD card.

closed 2019-04-16 by pavel-demin