pavel-demin/red-pitaya-notes #829: undefined reference to 'pthread_attr_init_'

ssooffiiaannee, 2019-04-17:
### Description of the setup:

 * Device: Red pitaya
 * SD card image:
 * Application:
 * Other relevant information: mcpha-server building

### Description of the problem:

### Steps to reproduce the problem:

When I try to build the mcpha-server as follow:

[image]

I get the following errors:
Undefined reference to 'pthread_attr_init'
Undefined reference to 'pthread_attr_setinheritsched' 
and more such errors

pavel-demin, 2019-04-17:
Thanks for testing these commands. I forgot to update them last time I updated the mcpha server code.

The first command should be
```
arm-linux-gnueabihf-gcc -static -O3 -march=armv7-a -mcpu=cortex-a9 -mtune=cortex-a9 -mfpu=neon -mfloat-abi=hard projects/mcpha/server/mcpha-server.c -o mcpha-server -lm -lpthread
```
with `-lm -lpthread` at the end.

I've just updated the mcpha notes.

ssooffiiaannee, 2019-04-17:
while waiting for the program to build, I'm trying to use the mcpha control program and I followed the steps, When I clic on connect I get the following error : cannot connect to 169.254.106.15:1001

pavel-demin, 2019-04-17:
Do you have the MCPHA application running on your Red Pitaya board?

ssooffiiaannee, 2019-04-17:
Yes, there is a led blinking, I guess it confirms the MCPHA application is in the SD?

pavel-demin, 2019-04-17:
When the first yellow LED is blinking, it's the LED blinker application that is running.

If you're using the `red-pitaya-alpine-3.9-armv7-20190215.zip` card image, then it contains several applications. The applications can be started via the web interface. Optionally, to start the mcpha applications automatically at boot time, you can copy its start.sh file from apps/mcpha to the topmost directory on the SD card. Otherwise, it's the LED blinker application that is started at boot time.

ssooffiiaannee, 2019-04-17:
Is it normal that these command are taking more than 3 hours to execute? Using one of the pre-built images, can I just execute the make command without re-executing " source helpers/build-all.sh ",

pavel-demin, 2019-04-17:
Yes, it takes the `build-all.sh` script several hours to run.

You can of course use any combination of the scripts and commands that you think is more suitable for your project.

Don't hesitate to edit the `build-all.sh` script and to remove all the projects that you don't need.

For example, if you want to build a SD card image with just the MCPHA application, then the following script should do the job:
```
source /opt/Xilinx/Vivado/2018.3/settings64.sh

make NAME=led_blinker all

make NAME=mcpha bit

sudo sh scripts/alpine-mcpha.sh
```

Since you aren't explaining what exactly you're trying to achieve, I can't help you more than this.

ssooffiiaannee, 2019-04-17:
I'm just trying to make the mcpha project work, then I'll look for how to modify it and make it work without rebuilding everything and waiting hours.

pavel-demin, 2019-04-17:
I think that you have already all the information for both of your goals.

It's of course possible to copy just the `mcpha.bit` and `mcpha-server` files to a already existing SD card.

The commands in my notes are just basic examples. They can be of course changed and combined in whatever way is more suitable for your project.

closed 2019-04-17 by pavel-demin