pavel-demin/red-pitaya-notes #830: Can't access the web application after loading the alpin linux

ssooffiiaannee, 2019-04-18:
### Description of the setup:

 * Device: Red Pitaya
 * SD card image:
 * Application: mcpha
 * Other relevant information:

### Description of the problem:

After copying the alpine linux into the SD card, when I try to run the web application using my browser and typing rp-fxxxxx.local 

### Steps to reproduce the problem:

 1.
 2.
 3.

alandong93, 2019-04-18:
I believe Pavel's Alpine Linux OS does not have a web interface like the original OS does.

pavel-demin, 2019-04-18:
@alandong93, my SD card images based on Alpine Linux do have a web interface.

@ssooffiiaannee, please provide more information. What SD card image are you using? What is your network setup?

If I use the [red-pitaya-alpine-3.9-armv7-20190215.zip](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20190215/red-pitaya-alpine-3.9-armv7-20190215.zip) SD card image, my web browser connects to rp-f0xxxx.local without any problem.

You can find more details about the SD card image based on Alpine Linux at [this link](http://pavel-demin.github.io/red-pitaya-notes/alpine/).

ssooffiiaannee, 2019-04-18:
I've put the zip file's content into the sd card, I did rp-f0xxxx into my web browser, nothing happens, the led is blinking on the red pitaya,

pavel-demin, 2019-04-18:
The information that you provide is still not sufficient for me to understand and reproduce the problem.
 
Are you using [red-pitaya-alpine-3.9-armv7-20190215.zip](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20190215/red-pitaya-alpine-3.9-armv7-20190215.zip) or your custom built SD card image?

How is the Red Pitaya board connected to your network? Do you have a router, a DHCP server?

If the rp-fxxxxx.local address doesn't work, maybe you could try using the IP address of the board?

If the image from the Red Pitaya website works for you, maybe the easiest solution would be to continue using this image. With this image, the MCPHA application can be installed via the application marketplace.

ssooffiiaannee, 2019-04-18:
I connected it directly to my pc, I can't find its ip address.
I'm using two sd cards, one with the image from red pitaya website, the other with the zip file you provided.

pavel-demin, 2019-04-18:
Please make an effort and try to answer all the questions from my previous comment and please do read what is already written in my notes at the following link:
http://pavel-demin.github.io/red-pitaya-notes/alpine/

ssooffiiaannee, 2019-04-18:
Now I'm using the OS provided by the RP website, I'm trying to download the mcpha project from application marketplace, but the RP needs access to the internet

 "RedPitaya is offline and cannot access the server!Please follow quick start instructions to put it online."  

and I don't know how to make the configurations. Even though I made the configurations in the RP website here : https://redpitaya.readthedocs.io/en/latest/quickStart/quickStart.html

I directly wired my RP to the PC, and I'm using windows 10

pavel-demin, 2019-04-18:
If you can't provide detailed answers to *all* my previous questions (the phrases with a question mark at the end), then I don't have enough information to help you.

ssooffiiaannee, 2019-04-18:
1-I'm using redpitaya from : https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html, I'll try to download the projects from application marketplace(as you said it may work).
2-The red pitaya is directly connected to my ethernet socket (to my PC).
3- In this case, I can access the RP with rp-0xxxx, but my problem is that the RP can't access internet

pavel-demin, 2019-04-18:
Thanks for your effort. However, I need more information about your network setup. Don't hesitate to provide as much details as you can. Do you have an Internet connection? If yes, then how it's done? Do you have a network router?

ssooffiiaannee, 2019-04-18:
Thank you, you have been of great help.
Yes I do have internet connection, yes I do have a network router, but I don't know what you mean by how it's done. When you try to upload that same image to the sd card, and plugging the RP into the pc, does it have internet connection immediatedly?

pavel-demin, 2019-04-18:
> yes I do have a network router,

Then why you don't connect both PC and Red Pitaya board to the router as it's shown in the [Red Pitaya documentation](https://redpitaya.readthedocs.io/en/latest/quickStart/first.html)?

>  I don't know what you mean by how it's done.

I mean how PC, Red Pitaya, network router and Internet are connected to each other and how the network interfaces are configured on your PC.

> When you try to upload that same image to the sd card, and plugging the RP into the pc, does it have internet connection immediately?

It depends on how the PC is configured. If I'm not mistaken, Windows 10 is by default configured to not provide the Internet connection to the devices connected to the Ethernet ports of the PC.

ssooffiiaannee, 2019-04-18:
The pc is connected by wifi, and I connected the RP to the router. I can't even access the web page by doing that.

pavel-demin, 2019-04-18:
This is strange.

Anyway, if the router doesn't work, then the only solution that I can think about is to connect Red Pitaya directly to your PC as you did before and to enable the Internet connection sharing as it's explained in [this article](http://www.informit.com/articles/article.aspx?p=2455390) or in many other articles that can be easily found with a web search engine.

Rolf-L, 2019-04-18:
Could be that the PCs IP-section is outside of the IP-adress of your RP?
this tool could help to figure out it:
https://www.softperfect.com/products/networkscanner/

ssooffiiaannee, 2019-04-18:
Thank your for all of your help, It turns out that I'm in a protected local network, my PC does not have network previleges to share network connection, so when I used another computer, it worked very well, sorry for all of the time I took from you. and again, thank you !!

closed 2019-04-19 by pavel-demin