pavel-demin/red-pitaya-notes #831: Problem of red_pitaya_0_92

chw177, 2019-04-19:
### Description of the setup:

 * Device: Redpitaya
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:
It might be a quite elementary problem since I am a starter of FPGA programming. I used VIVADO to generate the bitstream of the red_pitaya_0_92. After I copy the bitstream to the Redpitaya board, I type cat /root/xxxxx.bit > dev/xdevcfg. The Redpitaya terminal just stuck there. And the red light on Redpitaya is switched off immediately when I enter that command.

I searched on websites and found that maybe it's because I didn't instantiate the PS. I try to use the block_design.tcl to generate the PS. But it seems there are several bugs in that tcl file.

So how to exactly fix this problem?
### Steps to reproduce the problem:

 1. Add red_pitaya_0_92 to the new project and also include port.xdc and clock xdc as constraint files. Then do synthesis-> implement-> generate bitstream->get xxxx.bit
 2. scp the xxx.bit to Redpitaya. And run cat /root/xxxxx.bit > dev/xdevcfg.
 3. Redpitaya terminal stops there and red light on Redpitaya board is switched off.

pavel-demin, 2019-04-19:
Please try to fill all the fields of  in the beginning of the issue form. Which Red Pitaya board are you using 125-14 or 125-10? What is exactly the file name of the SD card image that is written to the SD card that you're using to boot your Red Pitaya board?

pavel-demin, 2019-04-19:
I've just tried to reproduce the problem with the following steps:
 - build red_pitaya_0_92.bit (`make NAME=red_pitaya_0_92 bit`)
 - boot Red Pitaya board using the latest version of the SD card image based on Alpine Linux ([red-pitaya-alpine-3.9-armv7-20190215.zip](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20190215/red-pitaya-alpine-3.9-armv7-20190215.zip))
 - copy red_pitaya_0_92.bit to Red Pitaya using scp
 - program the FPGA (`cat red_pitaya_0_92.bit > /dev/xdevcfg`)

After these steps the system is still running without any problem. The first yellow LED (LED0) is blinking.

I think that your system stops because there is a process running that communicates with the FPGA when you reprogramming it.

closed 2019-04-19 by pavel-demin

pavel-demin, 2019-04-19:
> I searched on websites and found that maybe it's because I didn't instantiate the PS. I try to use the block_design.tcl to generate the PS. But it seems there are several bugs in that tcl file.

I've just reread this part and realized that you're building my source codes without using my scripts.

I can only suggest to build red_pitaya_0_92.bit using the following make command:
```
make NAME=red_pitaya_0_92 bit
```

chw177, 2019-04-19:
Thanks Pavel. It works now.