pavel-demin/red-pitaya-notes #833: Is VNA working?

MxQL, 2019-04-20:
### Description of the setup:

 * Device: Red Pitaya VNA
 * SD card image:
 * Application: VNA
 * Other relevant information:

### Description of the problem:
I am trying the VNA software to retrieve data from a Redpitaya running VNA application. The experiment configuration is a transformer one: OUT1 excites the transformer primary winding, and IN1 is connected to the secondary winding of the transformer. In the past, I sometimes got meaningful data: a curve with resonance frequency. However, I am unable to get something correct this time, the VNA window is as attached. Every data at the cursor frequencies are NaN, while the connection with the Red Pitaya is successful. Are there something wrong in the setup? Could you help? 
Thanks a lot

Patrick
[image]

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-04-21:
Normally, the VNA application works. For example, it was confirmed in https://github.com/pavel-demin/red-pitaya-notes/issues/824.

Please provide more information. What SD card image are you using? What is the version of the VNA client program? If possible, could you please provide a cfg/ini file corresponding to this measurement?

MxQL, 2019-04-21:
Dear Pavel,

Thank you for your prompt reply.
I am not sure I know how to operate the VNA correctly.
Attached is the vna.ini file. What is strange is I set the start freq to 10 kHz, stop freq to 60 kHz, then save the ini file, but the values in the ini file are always those set by default (10 kHz, 6000 kHz).
I have measured the output signal at OUT1, there is no signal during the sweeping (AUTO button clicked). It seems that the board is not running (it runs correctly with the standard Redpitaya applications).
Thanks a lot for your help.
Patrick
[vna.ini.txt](https://github.com/pavel-demin/red-pitaya-notes/files/3101315/vna.ini.txt)

[image]

MxQL, 2019-04-21:
The VNA version is vna-win32-20190215
The SD card image is downloaded at the end of 2017. I already got it working after downloading it. I don't know how to get the version of the SD image. Attached is a screen copy of the SD card content.
[image]

pavel-demin, 2019-04-21:
Thanks for the screenshot. In https://github.com/pavel-demin/red-pitaya-notes/issues/659, your SD car image was red-pitaya-alpine-3.6-armhf-20171130.  From the dates in the screenshot, I'd say that it's still the same.

vna-win32-20190215 can't be used with red-pitaya-alpine-3.6-armhf-20171130. The versions of the SD card image and of the VNA client should be the same.

MxQL, 2019-04-21:
Dear Pavel,
Thank you very much for your help. It works now! 
Thanks again.
Patrick

closed 2019-04-21 by pavel-demin