pavel-demin/red-pitaya-notes #834: opt_design issues when using AXI BRAM Reader

mugundhan1, 2019-04-21:
### Description of the setup:

 * Device: Red pitaya 125-14
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:
When using AXI BRAM reader and connecting it to axi peripheral, the implementation during opt_design fails with : "[Opt 31-67] Problem: A LUT5 cell in the design is missing a connection on input pin I3, which is used by the LUT equation. This pin has either been left unconnected in the design or the connection was removed due to the trimming of unused logic. The LUT cell name is: design_1_i/ps7_0_axi_periph/xbar/inst/gen_sasd.crossbar_sasd_0/s_axi_bresp[0]_INST_0.
"

I tried the same design by just having the axi_config and axi_status registers, by removing the bram_reader core out, and it compiles fine. 
The error persists even when using the workaround suggested by xilinx i.e. to add this constraint: set_property DONT_TOUCH true [get_cells {design_1_i/ps7_0_axi_periph/xbar/inst/gen_sasd.crossbar_sasd_0/s_axi_bresp[0]_INST_0}]

Vivado version 2018.2

pavel-demin, 2019-04-21:
I use the AXI BRAM reader IP core in the MCPHA project:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/mcpha/hst.tcl#L35-L42

I regularly build the MCPHA project for all the releases of the SD card image and it builds without any problem. I've just checked the log files and I don't see any error message.

A similar problem is discussed in [AR 65764](https://www.xilinx.com/support/answers/65764.html). Maybe [AR 68238](https://www.xilinx.com/support/answers/68238.html) is also related to this problem.

I'd guess that you aren't using my scripts to build your project. If it's so, then I'd suggest to enable the same BD generation mode and the same optimization strategies that I enable in [my project.tcl script](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/project.tcl):
```
set_property synth_checkpoint_mode None [get_files $bd_path/system.bd]
```
```
set_property STRATEGY Flow_PerfOptimized_high [get_runs synth_1]
set_property STRATEGY Performance_NetDelay_high [get_runs impl_1]
```

closed 2019-04-21 by pavel-demin

mugundhan1, 2019-04-21:
Hi Pavel,

I used the instructions that you gave in the previous comments, and it worked. It compiles fine now!

Thanks for pointing this out !

Mugundhan