pavel-demin/red-pitaya-notes #835: How MCPHA project works?

ssooffiiaannee, 2019-04-21:
Dear Pavel, 

       Could you give me insights, documentations, about how the MCPHA project works?
I mean don't explain me how the shell scripts interract with each other, but how did you create those TCL files for example, an overview of the steps you went through to build such a project.

     Thank you

pavel-demin, 2019-04-21:
Could you please be more specific about what exactly is unclear to you? Have you read my notes and all the linked documents? Have you read the Xilinx user guides about the design flows and Tcl commands?

The design flows and Tcl commands are described in the following user guides:

https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug892-vivado-design-flows-overview.pdf
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug888-vivado-design-flows-overview-tutorial.pdf
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug896-vivado-ip.pdf
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug894-vivado-tcl-scripting.pdf
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug835-vivado-tcl-commands.pdf

A brief description of my Vivado work flow can be found in the LED blinker notes and in the slides of my presentation at Club Vivado 2016:

http://pavel-demin.github.io/red-pitaya-notes/led-blinker/
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1

ssooffiiaannee, 2019-04-21:
Sorry for any imprecise question, I'm trying to make it as clear as possible.
How did you create those tcl scripts? Did you start on vivado using IPs and then exported everything as tcl scripts?

pavel-demin, 2019-04-21:
Thanks for the more specific questions.

Please don't forget to answer two other questions from my previous comment. 

> How did you create those tcl scripts?

Using a text editor.

> Did you start on vivado using IPs and then exported everything as tcl scripts?

No.

ssooffiiaannee, 2019-04-21:
No I did not read everthing in Xilinx user guides about the design flows and Tcl commands, But I do know what it is about
I did read your notes and your documents.
I'm actually trying to understand all the mcpha project you did, and I'm thinking if I'm obliged to learn tcl commands to understand all this project?

pavel-demin, 2019-04-21:
OK. Since you've seen the slides linked from the LED blinker notes and from my first comment, then you should've probably seen the instructions on page 19 on how to analyze the project interactively. I think that it can be done without learning any Tcl command. If it doesn't work for you, then I'm afraid that I'm out of ideas.

closed 2019-04-22 by pavel-demin

ssooffiiaannee, 2019-04-22:
What is the difference between implementing the vivado project using IPs  and with tcl scripting?
Isn't using IPs easier?

pavel-demin, 2019-04-22:
> What is the difference between implementing the vivado project using IPs and with tcl scripting?
Isn't using IPs easier?

If you don't bother yourself with reading at least some of the linked documents, then why should I bother answering this kind of question? If you'd read them, you'd understand that this question doesn't make sense.

ssooffiiaannee, 2019-04-23:
I don't agree with you, my questions do make sense, maybee I formulated it badly. 
I read about the project mode and non-project mode, correct me If I'm wrong, but I think you did it using project mode. And there is a difference between these two modes.
What I'm asking, is wether or not I have to learn tcl scripting If I want to modify it and adapt it to my needs?

pavel-demin, 2019-04-23:
Have you checked the instructions on page 19 in [the slides](https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1) as I suggested in one of my previous comments? Have you tried to follow these instructions? If not, the please try the following:
- stop posting questions without providing me enough information to answer
- do try the instructions on page 19 in [the slides](https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1)
- explain me why this approach isn't what you're looking for

> I don't agree with you, my questions do make sense, maybee I formulated it badly.

If you really want to get clear answers to your questions, I think it's up to you to do your best and ask your questions as clearly as possible.

> What I'm asking, is wether or not I have to learn tcl scripting If I want to modify it and adapt it to my needs?

I'm afraid that you continue to answer questions without providing me enough information to answer.

Since I don't know anything about your needs/experience/background how do you expect me to answer this question?

ssooffiiaannee, 2019-04-23:
> Since I don't know anything about your needs/experience/background, then how do you expect me to answer this question? 
I'm actually an undergraduate student in electrical engineering who is working on your project, this project consists in understanding the mcpha project and tailoring it according to our needs.
> Have you checked the instructions on page 19 in the slides as I suggested in one of my previous comments? 
Yes, I did. I found it interesting, I'm still looking how to exploit that.
> explain me why this approach isn't what you're looking for
This approach may be the one I'm looking for, I don't know yet. I'm still working on it.