pavel-demin/red-pitaya-notes #836: Enhancement request/question: VNA operation below 500 kHz

glembler, 2019-04-22:
Hi Pavel,

I am investigating your VNA for calibration of radios at our local museum of radio, California Historical Radio Society, https://www.californiahistoricalradio.com/. Most consumer AM radios have an intermediate frequency of 455 kHz, which is just below the VNA 500 kHz lower frequency limit. Do you have any plans to add support for VNA operation below 500 kHz?

Thanks,

Gary Embler, N6RTA
gembler@google.com

pavel-demin, 2019-04-22:
Hi Gary,

I'm not sure if I understand where you've found this 500 kHz limit.

As far as I know, my VNA application for Red Pitaya seems to work at much lower than 500 kHz frequencies.

For example, the second image in my notes about my VNA application shows a Smith chart for the frequencies from 1 kHz to 10 MHz. Those measurements were done with a very simple series fixture.

Best regards,

Pavel

glembler, 2019-04-22:
Hi Pavel,

It's here: https://www.redpitaya.com/f172/vector-network-analyzer-module

*HARDWARE SPECIFICATIONS*
- freq. range 500k to 62MHz
   . . .

Thanks,

Gary

pavel-demin, 2019-04-22:
Thanks for the clarification.

The 500 kHz limit is specific to the hardware sold by the Red Pitaya company. So, the question about this limit should be addressed to the Red Pitaya company.

A very simple series fixture can be assembled from a few resistors and SMA or BNC connectors and it'll work at low frequencies without any problem.

For example, here is a picture of my setup:
[image]

pavel-demin, 2019-04-22:
Just found a couple of web pages showing very simple test fixtures:

http://www.scottyspectrumanalyzer.us/functioncrystal.html
https://destevez.net/2017/09/measuring-a-mains-choke-with-hermes-lite-vna/

Other solutions can be found in the documents linked from my notes about the VNA application.

closed 2019-04-22 by pavel-demin

glembler, 2019-04-26:
Thanks Pavel.

I sent a note to support@redpitaya.com asking about the limitation on the
low-frequency operation of their VNA bridge module. I haven't heard any
reply yet. Do you have a contact their who might know about it?

Gary

glembler, 2019-05-14:
Hi Pavel,

Just checking again - Would you happen to have a contact at Red Pitaya who can answer questions about their VNA bridge module?

Thanks,

Gary

pavel-demin, 2019-05-14:
If support@redpitaya.com doesn't answer, then I don't know who else can answer questions about the VNA bridge module.

glembler, 2019-06-25:
I finally heard from Nick at Red Pitaya. He shared a few specifications for
the VNA module. He also told me they are planning to do an improved version
of the VNA module. He did not have specifications or a schedule for the new
version.