pavel-demin/red-pitaya-notes #837: How to set a fix IP address for the Redpitaya running Pavel's Alpine image?

MxQL, 2019-04-23:
### Description of the setup:

 * Device:
 * SD card image: red-pitaya-alpine-3.9-armv7-20190215.zip
 * Application: VNA
 * Other relevant information:

### Description of the problem:
Hi,

Is it possible to get a fix IP address for the Redpitaya running the alpine image? I see that with the standard Redpitaya image, one can set a fix IP address. However, I don't know how to do that with the alpine image.

Thanks a lot

Maxim

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-04-23:
I think that the easiest is to configure it on your router.

If you really want to change the Alpine Linux configuration, then configuration of the IP addresses is in [/etc/dhcpcd.conf](https://github.com/pavel-demin/red-pitaya-notes/blob/master/alpine/etc/dhcpcd.conf).

I think that it should be enough to replace
```
interface eth0
fallback static_eth0
```
with
```
interface eth0
static ip_address=192.168.1.100/24
static routers=192.168.1.1
static domain_name_servers=192.168.1.1
```

The addresses should be of course replaced with the correct ones.

More details about the network configuration and other useful commands can be found at [this link](http://pavel-demin.github.io/red-pitaya-notes/alpine/#network-configuration).

MxQL, 2019-04-24:
Thank you, Pavel.

I just don't know where I have to put this file on the SD card. The folder etc doesn't exist in the standard red-pitaya-alpine-3.9-armv7-20190215. Have I to add the etc folder to the SD card?
[image]

Maxim

pavel-demin, 2019-04-24:
Have you tried to configure your router? It's much easier.

Or if the standard Red Pitaya image works better for you, then the next easiest solution is to use the standard Red Pitaya image and install the VNA application via the application marketplace.

> I just don't know where I have to put this file on the SD card

The /etc/dhcpcd.conf configuration file should be edited on running Red Pitaya.

Here are some link with instructions:

- two solutions to connect to the Red Pitaya command line
https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html
https://redpitaya.readthedocs.io/en/latest/developerGuide/os/ssh/ssh.html

- some useful Alpine Linux commands
http://pavel-demin.github.io/red-pitaya-notes/alpine/

Once you are connected to the Red Pitaya command line, the commands to edit and to save configuration changes to SD card are the following:

```
# edit /etc/dhcpcd.conf using nano editor
nano /etc/dhcpcd.conf

# save configuration changes to SD card
lbu commit -d
```

If these instructions don't work for you, then I suggest to try the two easier solutions that I've proposed in the beginning of this comment.

closed 2019-04-26 by pavel-demin