pavel-demin/red-pitaya-notes #838: aquistion with zoom?

jim-martz, 2019-04-26:
Hello,
I am interested in a project using the Red Pitaya with a larger buffer
and being able to read the buffer with SCPI using selectable decimation.
I have seen your MCPHA project and adc_recorder_trigger, maybe you 
can give me some advice on what all may be involved and how your projects
may be able to help me?

Here is a list of what my project needs to do:
1. Triger an analog acquisition either by a voltage level or external trigger. 
2. Once triggered write the data at 125Msps to a 2MS buffer in DDR memory, for each channel.
3. After the data is stored be able to read it at a 128 decimation, through TCP.
4. Then be able to read selectable sections of the raw data with no decimation, or even better 
     selectable decimation. (Think of this like a scope with a zoom function.)

Another option that may be easier to implement would be to have 2 buffers for each channel.
One that is a 2MS buffer that holds all the samples and can be read at any location.
And a second smaller buffer that stores the data with 128 decimation.
 
The control of this would need to be with SCPI functions over the Ethernet, the end software
would be LabView.    

I have some experience with Linux and C programing, but little experience with fpga programing.
I am trying to figure out what all pieces of my project don’t currently exist and try to determine how difficult this project would be for me and try to determine how long it may take. 

One other question, is there a guide page for adc_recorder_trigger like there is for MCPHA?

Thanks,
Jim

### Description of the setup:

 * Device:
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-04-26:
I don't think that I can help much. I'd say that the MCPHA project has almost all the parts needed for your project. They just need to be rewired.

> Another option that may be easier to implement would be to have 2 buffers for each channel.
One that is a 2MS buffer that holds all the samples and can be read at any location.
And a second smaller buffer that stores the data with 128 decimation.

I agree that this seems easier to implement.

I think that it would be even easier to implement only the 2 MS buffer. The zooming part could run on the PC.

> One other question, is there a guide page for adc_recorder_trigger like there is for MCPHA?

I don't have any guide page for the adc_recorder_trigger project. It's just a proof of concept project.

The MCPHA application contains most of the adc_recorder_trigger functionality and it has better usability.

jim-martz, 2019-04-26:
Thank you for the information.
My initial thinking was to implement the zoom on the PC end, but
then I realized how long it would take to send 2Ms buffers over Ethernet.
I am hoping to do up to 4 captures per second. With 2Ms I could not even send
1 capture of data in a second. 

For the most part 128 decimation would be fine, but there are parts of the waveform,
particularity the front end that I need the 125Msps resolution.
Next week I am going to try to use one channel at 125Msps to get the front end,
and the second channel at 128 decimation, each with the default 16k buffer. 
This may just work for my project, but would mean I need to use twice as many red pitayas.

At some point after I get some more understanding of all the programing, I will look at
the MCPHA project some more. I also seen the DAQServer project and posted there,
it may also be something that may work for me.

Thanks,
Jim

BeauWebber, 2019-04-26:
I have a related data capture need :
For some NMR sequences, there is ideally a need to sample just the peaks of an extended sequence of pulses.
I am currently sampling continuously, and then selecting the peaks in the PC and polynomial fitting them.

However I have a third option, as I am running the high-level language (Apl) in a split mode,
with the Apl Server running in the Linux processor, and the Apl Client on the host desktop displaying and filing the captured data.
What I have is an C program in the Linux processor, talking to the firmware, and feeding the data to the Apl Server.
At the moment I am passing all the captured data to the client, but I have the third option of processing the data in the C code / Apl Server.

Is something like this an option for you, Jim ?
cheers,
                Beau Webber

-- Kentforlife.net - the email service for alumni of the University of Kent

jim-martz, 2019-04-29:
Beau,

Thank you for the reply.
I will keep this in mind, there may be a way to have the Linux processor limit
the amount of data that is sent, or select the data that is sent.

Thanks,
Jim

closed 2019-05-14 by pavel-demin