pavel-demin/red-pitaya-notes #843: Ports constraints

ssooffiiaannee, 2019-05-10:
Hi Pavel, 
    Could you tell me what are these pins for? They are not connected to any IP in vivado, This is the MCPHA project, 
   
    And also in the cfg/ports.xdc file, where are Vaux0_v_p and others coming from ? I don't find them in any tcl script, I found only  Vaux0 Vaux1 Vaux8 Vaux9 Vp_Vn in cfg/ports.tcl.
    

[image]
[image]
[image]

pavel-demin, 2019-05-10:
The Vaux pins are the XADC inputs that are connected to the analog input pins of the [extension connector E1](https://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extent.html#extension-connector-e2). The XADC is used in the SDR transceiver compatible with HPSDR.

According to the [Red Pitaya schematics](https://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/shem.html), the Vp_Vn pins are connected to the power supply voltage monitor.

closed 2019-05-10 by pavel-demin

ssooffiiaannee, 2019-05-12:
How long did you take to accomplish the mcpha project? Did you do it alone?

pavel-demin, 2019-05-12:
> How long did you take to accomplish the mcpha project? Did you do it alone?

Please stop abusing this issue tracker by posting meaningless questions.

It's a hobby project. Working on this project is supposed to be fun and stress free. Tracking time isn't my idea of stress-free fun.

It's not a project made of stone and stick in the prehistoric times. Similarly to a vast majority of projects done in today's world, it's based on a lot of work done by a lot of people.