pavel-demin/red-pitaya-notes #844: Development Machine

sixtemesseven, 2019-05-18:
After pulling my hair out with windows and giving up compiling from source for a couple weeks I tried to install your virtual setup described here:
http://pavel-demin.github.io/red-pitaya-notes/development-machine/

This seems to work well until the point:
--Installing Vivado Design Suite

I have downloaded the installer package but I'm not able to transfer it. My idea was to create a shared folder but it seems I'm not able to install the guest addition tools. Some answers I found said that it cannot be installed in Debian 9.x ...(?) I different tutorials but my modprobe always returns package not found :(

Anyway, do you have a specific reason why it should be a headless system? I mean it is a good idea if you have a laptop and want to run it on a powerfull second machine but I do not have that problem. So I would much prefer to have a GUI... The hole ssh / x11 is yet another complication it seems to me. Please correct me if I'm wrong.

I'm not used to Debian at all and have minimal expirience in Linux in general. So a detailed step by step solution would be absolutly terrific :)

pavel-demin, 2019-05-18:
I've updated the Development Machine notes a few days ago to install Debian 9 and I tested the instructions after the update. They work for me without any problem.

I'm using wget to download Xilinx_Vivado_SDK_2018.3_1207_2324.tar.gz to the virtual machine. I start to download Xilinx_Vivado_SDK_2018.3_1207_2324.tar.gz in a web browser, then I stop the transfer and copy the download link to the virtual machine command line.

It's also possible to download Xilinx_Vivado_SDK_2018.3_1207_2324.tar.gz to a different computer and then use scp to copy it to the virtual machine.

If the resulting virtual machine doesn't match your requirements, then I'm afraid I can't help you.

closed 2019-05-18 by pavel-demin

pavel-demin, 2019-05-18:
If you have Windows 10, then maybe installing [Ubuntu](https://www.microsoft.com/en-us/p/ubuntu/9nblggh4msv6) from the app store could be a better solution.

BeauWebber, 2019-05-18:
Ah thanks, I will investigate – I am facing the demise of Windows 7 - Beau

-- Kentforlife.net - the email service for alumni of the University of Kent

pavel-demin, 2019-05-18:
> I'm not able to install the guest addition tools

I've just checked that the virtualbox-guest-dkms is now in the stretch-backports repository. Here are the commands to enable this repository and to install virtualbox-guest-dkms:
```
su -
printf "\ndeb http://deb.debian.org/debian stretch-backports main contrib non-free\ndeb-src http://deb.debian.org/debian stretch-backports main contrib non-free\n" >> /etc/apt/sources.list
apt-get update
apt-get install virtualbox-guest-dkms
```
I've just added these commands to the installation script.

sixtemesseven, 2019-05-18:
> I've updated the Development Machine notes a few days ago to install Debian 9 and I tested the instructions after the update. They work for me without any problem.
>
> I'm using wget to download [...]

Hi Pavel

Thank you for your answers and the updated installation script. I will try again.

Sorry, I did not mean to disagree with your setup. I was mainly wondering why one would make things more complicated.

I had no idea of wget and it's function. As I said, not used to Linux.

Thank you once more
Andreas

pavel-demin, 2019-05-19:
> So I would much prefer to have a GUI...

It's possible to install a relatively compact but usable desktop environment with the following commands:
```
su -
apt-get install virtualbox-guest-x11 xfce4 xfce4-terminal
```
It starts automatically after rebooting the virtual machine.

sixtemesseven, 2019-05-21:
> > So I would much prefer to have a GUI...
> 
> It's possible to install a relatively compact but usable desktop environment with the following commands:
> 
> ```
> su -
> apt-get install virtualbox-guest-x11 xfce4 xfce4-terminal
> ```
> 
> It starts automatically after rebooting the virtual machine.

This works, thank you! 
Interestingly the virtualbox tools still do not work (copy/paste or drag&drop).
If I install the tools from the .iso promted the gui fails to start... 

How do you get past the login at the xilinx page with wget? I can copy the download link but it will load a html file and not the bin file.

sixtemesseven, 2019-05-21:
Nevermind my last comment. I was able to install chrome on the gui and download the file via chrome to opt/Xilinx

pavel-demin, 2019-05-21:
> Interestingly the virtualbox tools still do not work (copy/paste or drag&drop).

I think that the clipboard sharing should be enabled via the Devices menu.

> I can copy the download link but it will load a html file and not the bin file.

I start to download Xilinx_Vivado_SDK_2018.3_1207_2324.tar.gz in a web browser, then I stop the transfer and copy the download link from the browser's downloads manager.

BeauWebber, 2019-05-22:
Linux – Ubuntu on Windows 10
Comments re installing Ubuntu and Xilinx_Vivado. – Dr Beau Webber.
Pavel Demin said :
“If you have Windows 10, then maybe installing Ubuntu<https://www.microsoft.com/en-us/p/ubuntu/9nblggh4msv6> from the app store could be a better solution”
So I gave it a try :
The instructions for enabling the Windows Linux were fine, see the overview tab : https://www.microsoft.com/en-us/p/ubuntu/9nblggh4msv6?activetab=pivot:overviewtab
The Linux Virtual Machine installation is at :
C:\Users\<win user>\VirtualBox VMs\Linux – Ubuntu
The text desktop one gets has a little red icon in the top left corner, where lots of useful things are hidden, including copy and paste, and settings.
I found the default colours mostly unreadable, but in settings one can change them.

The 19GB Xilinx_Vivado  installation then goes (see below) at :
C:\Users\<win user>\AppData\Local\Packages\CanonicalGroupLimited.UbuntuonWindows_xxx
This is not good for me as C:\ is only a small 100 GB SSD meant for the Windows programs.

I have been installed as a simple Linux user so all operations in /opt have to be prefixed with sudo.
I know about su, but it has not yet worked for me, perhaps I have forgotten to do something.
From Linux, the Windows drives are found at :
/mnt/c
/mnt/d
etc.
So I first downloaded the 19GB  Zilinx_Vivado  .tar.gz  file to space on my d:\ drive, and then accessed it from Linux.
I followed Pavel Demin ‘s  :
http://pavel-demin.github.io/red-pitaya-notes/development-machine/
starting at :

Installing Vivado Design Suite

  *   Download “Vivado HLx 2018.3: All OS installer Single-File Download” from the Xilinx download page<https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vivado-design-tools/2018-3.html> (the file name is Xilinx_Vivado_SDK_2018.3_1207_2324.tar.gz)
  *   Create the /opt/Xilinx directory, unpack the installer and run it:
mkdir /opt/Xilinx
cd /opt/Xilinx
tar -zxf Xilinx_Vivado_SDK_2018.3_1207_2324.tar.gz
cd Xilinx_Vivado_SDK_2018.3_1207_2324
sed -i '/uname -i/s/ -i/ -m/' xsetup
./xsetup

Because of the limited space on the C:\ drive, there was not room to copy the compressed file to the /opt /Xilinx directory, and then also unpack it.  I actually had to cd to /op/Xilinx and then say :
tar -zxf /mnt/d/<win path>/Xilinx_Vivado_SDK_2018.3_1207_2324.tar.gz

Xilinx_Vivado seems then to configure fine, but with ./xsetup where I think it is looking for an Xwindows display :
Exception in thread "main" java.lang.UnsatisfiedLinkError: /opt/Xilinx/Xilinx_Vivado_SDK_2018.3_1207_2324/tps/lnx64/jre9.0.4/lib/libawt_xawt.so: libXrender.so.1: cannot open shared object file: No such file or directory

I have tried installing virtualbox-guest-x11, but none of the packages are found.
I found the following web site, which seems competent :
https://www.makeuseof.com/tag/linux-desktop-windows-subsystem/ :

·         It suggests :

Start by running an update and upgrading Ubuntu:

sudo apt update

sudo apt upgrade

·         (this takes ages, but seems worthwhile to get a fully updated system)
While this upgrade is running, head to Sourceforge to download and install the VcXsrv Windows X Server utility<https://sourceforge.net/projects/vcxsrv/files/latest/download>. (Other X Servers are available for Windows, including Xming<https://sourceforge.net/projects/xming/>and MobaXterm<https://mobaxterm.mobatek.net/>. For the remainder of this guide, we’ll be using VcXsrv.)

·         (I had already installed Xming, so I skipped this section.
Note this is a native Windows version, and installs on the Windows side.
I then continued with : )

Many Linux desktop environments<https://www.makeuseof.com/tag/best-linux-desktop-environments/> (LDEs) are available. We’re going to keep things simple and install a lightweight environment called LXDE. To install, input:

sudo apt install lxde

·         (this again takes ages)

Following installation of LXDE, input this command

export DISPLAY=:0

export LIBGL_ALWAYS_INDIRECT=1

startlxde
and bingo a desktop, with the usual Linux type bar along the bottom, and on which one can have Windows windows, such a file listings, or web email windows, as well as Linux consoles etc. I could get to like this ….
Copy and paste from Windows text to Linux text works fine.

OK, to return to Xilinx-Vivado :

cd /opt/Xilinx
cd Xilinx_Vivado_SDK_2018.3_1207_2324
sed -i '/uname -i/s/ -i/ -m/' xsetup
./xsetup
(sudo now not needed it seems)
Xilinx splash-screen appears, and after a pause, a series of configuration windows appear.

The first few are OK, then it says :
“Select Edition to Install”.
Pavel’s guide says :
Follow the installation wizard and don’t forget to select “Software Development Kit” on the installation customization page (for detailed information on installation, see UG973<https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug973-vivado-release-notes-install-license.pdf>)
I am afraid I get totally lost at this point. Help Please !!
Cheers,
                Beau

pavel-demin, 2019-05-22:
> “Select Edition to Install”.

It should be OK to select WebPACK.

BeauWebber, 2019-05-23:
Thanks, sorry, just seen this.
Proceeds OK until it says I will not have enough disk space to install.
Major rethink now.

pavel-demin, 2019-05-24:
> libXrender.so.1: cannot open shared object file: No such file or directory

This error message indicates that the libxrender1 package is missing.

The solution is to install all the packages listed in the [pre-requirements section](http://pavel-demin.github.io/red-pitaya-notes/led-blinker/#pre-requirements) of the LED blinker notes.

The [installation script](https://github.com/pavel-demin/red-pitaya-notes/blob/gh-pages/etc/debian.seed#L81-L89) from the development machine notes installs these packages automatically.