pavel-demin/red-pitaya-notes #845: adc-recorder-trigger recording/reading samples incorrectly

IainMckeeman, 2019-05-21:
### Description of the setup:

 * Device: red pitaya
 * SD card image: latest alpine with pre built apps
 * Application: adc-recorder-trigger
 * Other relevant information: 

### Description of the problem:
Hi Pavel,

Thanks for developing all these applications and projects, they are very useful to learn from.

I am currently looking at the adc-recorder-trigger project. I have generated the bitstream in Vivado after building the project with your helper and have successfully deployed it on the red pitaya. I compiled adc-recorder-trigger.c on the red pitaya and had these warnings:

rp-f05a91:~# gcc -o adc-trig adc-recorder-trigger.c
adc-recorder-trigger.c: In function 'main':
adc-recorder-trigger.c:83:10: warning: passing argument 1 of 'munmap' discards  volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
   munmap(cfg, sysconf(_SC_PAGESIZE));
          ^~~
In file included from adc-recorder-trigger.c:4:
/usr/include/sys/mman.h:111:13: note: expected 'void *' but argument is of type 'volatile void *'
 int munmap (void *, size_t);
             ^~~~~~
adc-recorder-trigger.c:84:10: warning: passing argument 1 of 'munmap' discards  volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
   munmap(sts, sysconf(_SC_PAGESIZE));
          ^~~
In file included from adc-recorder-trigger.c:4:
/usr/include/sys/mman.h:111:13: note: expected 'void *' but argument is of type 'volatile void *'
 int munmap (void *, size_t);
             ^~~~~~
adc-recorder-trigger.c:85:10: warning: passing argument 1 of 'munmap' discards  volatile' qualifier from pointer target type [-Wdiscarded-qualifiers]
   munmap(ram, sysconf(_SC_PAGESIZE));
          ^~~
In file included from adc-recorder-trigger.c:4:
/usr/include/sys/mman.h:111:13: note: expected 'void *' but argument is of type 'volatile void *'
 int munmap (void *, size_t);
             ^~~~~~

However the c-code seems to work as it should. 

My goal is to very simply record one sample on an external trigger and write the sample to the sd card. In the final application this would be at 4kHz. I have been testing by inputing a 1Hz sine wave into IN1 and a trigger of 10 Hz and recording 20 samples. I should see two periods of sine wave but the samples recorded look something like this:

[image]

I have changed in adc-recorder-trigger.c...
number of samples before trigger =0
trigger edge = 1 (positive)
total num samples = 2
and added a for loop to read 20 samples here (with usleep(100)):

for(j=0; j<20; j++)
  {
  /* start oscilloscope */
  *((uint16_t *)(cfg + 0)) |= 2;
  *((uint16_t *)(cfg + 0)) &= ~2;

  /* wait when oscilloscope stops */
  while(*((uint32_t *)(sts + 0)) & 1)
  {
    usleep(100);
  }

  start = *((uint32_t *)(sts + 0)) >> 1;
  start = (start  ) & 0x007FFFC0;

  /* print IN1 and IN2 samples */
  for(i = 0; i < 1; ++i)
  {
    offset = ((start + i) & 0x007FFFFF) * 4;
    value[0] = *((int16_t *)(ram + offset + 0));
    //value[1] = *((int16_t *)(ram + offset + 2));
    //printf("%5d %5d\n", value[0]);//, value[1]);
    printf("%5d \n", value[0]);
  }

}

I am sure this problem is due to my limited C-code experience but was wondering if you could help at all.

Thanks.

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-05-21:
I think that the code resetting oscilloscope and ram writer should be inside the loop.

IainMckeeman, 2019-05-21:
Your suggestion worked when I increased the samples read to 256, I can just record the first sample after the trigger. 
Thanks for your help

closed 2019-05-21 by pavel-demin

falcon98, 2019-06-06:
Hi

falcon98, 2019-06-06:
Hi Pavel,

Hope you are doing good. I had a requirement of saving multi triggered data form a laser system. The trigger rate is 10 Hz and I need to save 256k samples in total. 
Do you think that it will be possible from this program? I tried to make some changes but I am not able to go beyond 1 Hz trigger rate.
Following is the code:
`while(count<5)
{
  /* wait when oscilloscope stops */
  while(*((uint32_t *)(sts + 0)) & 1)
  {
    usleep(1000);
  }
printf("Oscilloscope completed \n");
  start = *((uint32_t *)(sts + 0)) >> 1;
  start = (start - 256*1*1024) & 0x007FFFC0;

printf("Print of samples started \n");
double time_spent;
clock_t begin = clock();
  /* print IN1 and IN2 samples*/
 for(i = 0; i < 512*1* 1024; ++i)
  {
    offset = ((start + i) & 0x007FFFFF) * 4;
    value[0] = *((int16_t *)(ram + offset + 0));
    value[1] = *((int16_t *)(ram + offset + 2));
  // printf("%5d %5d\n", value[i]);
fprintf(fPointer, "%d \n", value[0]);
fprintf(fPointer, "%d \n", value[1]);

  }    
 // commented entire loop for filw writing
//offset = ((start ) & 0x007FFFFF) * 4;
//value[0] = *((int16_t *)(ram + offset + 0));
 

//offset = ((start ) & 0x007FFFFF) * 4;

//fwrite(ram+offset, sizeof(int16_t),2*1024 * 1024,fPointer);

//fwrite(value, sizeof(value),1,fPointer);

//int size_value;
//printf("\n size of array: %d\n",size_value);

clock_t end = clock();
time_spent = (double)(end-begin)/ CLOCKS_PER_SEC;
printf("Time spent: %f", time_spent);

count = count +1;
}

`

I tried with both binary and text files. Can you suggest some better optimization. Is is possible to save the file for locally at this event rate.
THanks

pavel-demin, 2019-06-06:
> The trigger rate is 10 Hz and I need to save 256k samples in total.

If I'm not mistaken, it corresponds to 10 MB/s (256 * 2^10 * 10 * 4 / 2^20). If you're writing the data to a SD card, then I'd suggest to check if the Red Pitaya board can write data to this SD card at this speed.

> I tried with both binary and text files. Can you suggest some better optimization.

I think that writing binary files using `fwrite` is the right solution. I'd try something like the following **untested** code:

```
#include <stdio.h>
#include <stdint.h>
#include <string.h>
#include <unistd.h>
#include <sys/mman.h>
#include <fcntl.h>

int main()
{
  FILE *output;
  int fd, i;
  uint32_t start, total;
  volatile uint32_t *slcr, *axi_hp0;
  volatile void *cfg, *sts;
  void *ram, *buf;

  if((fd = open("/dev/mem", O_RDWR)) < 0)
  {
    perror("open");
    return 1;
  }

  slcr = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0xF8000000);
  axi_hp0 = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0xF8008000);
  cfg = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0x40000000);
  sts = mmap(NULL, sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0x40001000);
  ram = mmap(NULL, 8192*sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED, fd, 0x1E000000);
  buf = mmap(NULL, 8192*sysconf(_SC_PAGESIZE), PROT_READ|PROT_WRITE, MAP_SHARED|MAP_ANONYMOUS, -1, 0);

  /* set HP0 bus width to 64 bits */
  slcr[2] = 0xDF0D;
  slcr[144] = 0;
  axi_hp0[0] &= ~1;
  axi_hp0[5] &= ~1;

  /* configure trigger edge (0 for negative, 1 for positive) */
  *((uint16_t *)(cfg + 2)) = 0;

  /* set trigger mask */
  *((uint16_t *)(cfg + 4)) = 1;

  /* set trigger level */
  *((uint16_t *)(cfg + 6)) = 1;

  /* set number of samples before trigger */
  *((uint32_t *)(cfg + 8)) = 1024 - 1;

  /* set total number of samples (up to 8 * 1024 * 1024 - 1) */
  total = 262144;
  *((uint32_t *)(cfg + 12)) = total - 1;

  /* set decimation factor for CIC filter (from 5 to 3125) */
  /* combined (CIC and FIR) decimation factor is twice greater */
  *((uint16_t *)(cfg + 16)) = 5;

  output = fopen("output.dat", "wb");

  for(i = 0; i < 10; ++i)
  {
    /* reset oscilloscope and ram writer */
    *((uint16_t *)(cfg + 0)) &= ~1;
    *((uint16_t *)(cfg + 0)) |= 1;

    /* start oscilloscope */
    *((uint16_t *)(cfg + 0)) |= 2;
    *((uint16_t *)(cfg + 0)) &= ~2;

    /* wait when oscilloscope stops */
    while(*((uint32_t *)(sts + 0)) & 1)
    {
      usleep(100);
    }

    start = *((uint32_t *)(sts + 0)) >> 1;
    start = (start - 1024) & 0x007FFFFF;

    if(start + total <= 0x007FFFFF)
    {
      memcpy(buf, ram + start * 4, total * 4);
      fwrite(buf, 1, total * 4, output);
    }
    else
    {
      memcpy(buf, ram + start * 4, (0x007FFFFF - start) * 4);
      fwrite(buf, 1, (0x007FFFFF - start) * 4, output);

      memcpy(buf, ram, (start + total - 0x007FFFFF) * 4);
      fwrite(buf, 1, (start + total - 0x007FFFFF) * 4, output);
    }
  }

  fclose(output);

  return 0;
}
```

falcon98, 2019-06-07:
Hi Pavel,

thanks for your guidance , it worked for 10 Hz rate at 256 K samples and also till 512K Samples.

The data streaming rate should be  (256K (samples)*2(Bytes/Samples)*10 = 5MBps.Since 256K samples are transferred every 100ms.

falcon98, 2019-07-23:
Hi Pavel,

I was trying to understand things in detail and had a doubt. For most of these projects, HP0 port is used to transfer data from PL to DDR3 controller. From here the address space is read using C program.

What will be the benefit in using DMA block in these projects. HP ports can provide data rate upto 1200 MBPS as per xilinx documentation. 
And in these projects the maximum data rate being targeted is 125MSPS*2(S/B)*2CH = 1000MBPS so HP port is able to withstand the data rate.

Pls let me know if my understanding is correct or not. What will be the advantages/complications in using DMA and the reason for using HP0 port is that  the data rates fall under HP0 port capability .

pavel-demin, 2019-07-23:
> What will be the benefit in using DMA block in these projects.

I'm not sure if I understand the question. Transferring data from PL to DDR3 is already direct memory access (DMA).

> What will be the advantages/complications in using DMA and the reason for using HP0 port is that the data rates fall under HP0 port capability.

The advantages/complications with respect to what?

The adc_recorder_trigger project already uses both DMA and HP0.

falcon98, 2019-07-24:
In the project the memory writer IP is connected to HP0 port of PS. I dont see any DMA IP in the code. Thats why I thought that the data is being transferred to DDR via HP0 port.
How is DMA being used in this case. Or you mean that implicitly DMA usage is there when we use HP0 interface.

I thought that for high speed data transfer we have options available directly use HP0 or use DMA via HP0

pavel-demin, 2019-07-24:
> I dont see any DMA IP in the code. 

DMA means direct memory access.  The memory writer directly accesses the memory without using the CPU. So, the memory writer module implements DMA.

I once wrote a long post about DMA on the Red Pitaya forum:
https://forum.redpitaya.com/viewtopic.php?t=551&p=2200#p2200

pavel-demin, 2019-07-25:
I've just realized that there is something wrong with the calculations from one of your previous comments.

> HP ports can provide data rate upto 1200 MBPS as per xilinx documentation.

When configured as 64-bit and clocked at 125 MHz, the data rate of the HP port should be around 7630 Mbits/s (64 * 125 * 10^6 / 2^20) or 954 Mbytes/s.

> And in these projects the maximum data rate being targeted is 125MSPS*2(S/B)*2CH = 1000MBPS

The data rate from the dual channel ADC is around 3815 Mbits/s (32 * 125 * 10^6 / 2^20) or 477 Mbytes/s.