pavel-demin/red-pitaya-notes #846: Using pulse generator

ssooffiiaannee, 2019-05-23:
### Description of the setup:

 * Device: Red Pitaya
 * SD card image: Provided by RedPitaya, this one : https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html
 * Application: MCPHA, pulse generator
 * Other relevant information:

### Description of the problem:
I started the mcpha application, the histogram works perfectly, but I don't know how to make the pulse generator work.

### Steps to reproduce the problem:

 1.Enter in the RP web interface.
 2.Lunch mcpha application
 3.Connect the RP to the mcpha interface,
Histograms work perfectly , but not the pulse generator, I tried to generate pulses (like the name points out) to an external oscilloscope, but no impulse is generated.

[image]

I hope it's clear !!

pavel-demin, 2019-05-23:
The pulse generator requires an input spectrum histogram that should be loaded using the 'Read file' button. The input spectrum histogram file should be a text file with 4096 lines with each line containing a number of entries in the corresponding histogram bin.

Here is an example of input file:
[test.zip](https://github.com/pavel-demin/red-pitaya-notes/files/3211698/test.zip)

ssooffiiaannee, 2019-05-23:
Thanks, So I can plug the IN1 to the Out1 of the same RP, generate the impulses, and check whether we will have the same spectrum?

pavel-demin, 2019-05-23:
Yes, it's possible to obtain more or less similar input and output spectra.

When I tested it last time, it looked like the following:

[image]

The input spectrum is blue, the output spectrum is black.

closed 2019-05-23 by pavel-demin