pavel-demin/red-pitaya-notes #847: gcc 6.0 for u-boot compilation

waqarrashid33, 2019-05-24:
I am been trying to run some of your examples on my ubuntu 16.04 machine. But when it reached the u-boot compilation it gives an error that it needs gcc verison 6.0.

Where do I get it?
Before this error it wasn't able to find arm-linux-gnueabihf-gcc but then I installed using apt install. now it says the version is not matching. I looked on the internet for solutions but I couldn't understand much of them.

waqarrashid33, 2019-05-24:
It seems to be compiling on 2017.2

waqarrashid33, 2019-05-24:
That results in newer bsp verison and pathches are for older ones. I don't know if there are differences between bsp version 6_3 and 5_0

pavel-demin, 2019-05-24:
I'm using Vivado 2018.3 and the corresponding u-boot version from the Xilinx repository:

https://github.com/Xilinx/u-boot-xlnx/tree/xilinx-v2018.3

The GCC compiler comes with Vivado 2018.3. The GCC compiler version is 7.3.1.

If you're going to use my scripts, then I'd suggest to install Vivado 2018.3.

pavel-demin, 2019-05-24:
Here is a link to the last commit before I switched to Vivado 2015.2:

https://github.com/pavel-demin/red-pitaya-notes/tree/a90f33c675edc5041ea42841d0415937aae3c96a

If I'm not mistaken, this version should still work with Vivado 2015.1.

waqarrashid33, 2019-05-24:
Thanks. I was looking for the 2015.1 point.
Why aren't there any patches for my version of bsp?

pavel-demin, 2019-05-24:
> Why aren't there any patches for my version of bsp?

I'm afraid that I don't understand this question. What is BSP? What patches are you looking at?

closed 2019-05-25 by pavel-demin

waqarrashid33, 2019-05-26:
I tried it with 2018.3 and everything seems working.
By patches for BSP I meant patches/standalone_v5_0.patch as it was trying to apply this patch but couldn't find the directory as it was named standalone_v6_3, but it didn't gave any error with 2018.3. Thanks for your help.

pavel-demin, 2019-05-26:
Thanks for the clarification.

The bare_metal_test project was just a proof of concept that I once tried four years ago just out of curiosity. Currently, I have no motivation to maintain it up-to-date.

waqarrashid33, 2019-05-26:
ok. I am trying to make it work as I am unable to properly configure remoteproc on RedPitaya. Due to some reason the example modules have compilation errors and at the moment I don't know how to fix them. I guess I would have to go to petalinux on Zedboard just to wrap my head around some of the key concepts, rather than trying to fix something I don't know how it works in the first place. Thanks for the help.

Just one question though, do you mind of I post some of my findings related to your bare_metal_test here in this issue?

UPDATE:
Without the applying the patches I am getting the errors without applying the above mentioned patch:
```
make[1]: Entering directory '/home/waqar/workspace/pavel-demin/red-pitaya-notes/tmp/bare_metal_test.cpu1/app_cpu1_bsp/ps7_cortexa9_1/libsrc/standalone_v6_8/src'
Compiling standalone
/tmp/ccTLklKP.s: Assembler messages:
/tmp/ccTLklKP.s:69: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:73: Error: selected processor does not support `isb' in ARM mode
/tmp/ccTLklKP.s:135: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:139: Error: selected processor does not support `isb' in ARM mode
In file included from xtime_l.c:49:0:
xtime_l.h:87:9: note: #pragma message: For the sleep routines, Global timer is being used
 #pragma message ("For the sleep routines, Global timer is being used")
         ^~~~~~~
In file included from sleep.c:53:0:
xtime_l.h:87:9: note: #pragma message: For the sleep routines, Global timer is being used
 #pragma message ("For the sleep routines, Global timer is being used")
         ^~~~~~~
/tmp/ccTLklKP.s: Assembler messages:
/tmp/ccTLklKP.s:73: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:165: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:198: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:228: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:299: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:385: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:456: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:554: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:653: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:729: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:825: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:934: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:1042: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:1270: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:1355: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:1506: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:1699: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:1784: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:1832: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:1944: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:1987: Error: selected processor does not support `dsb' in ARM mode
/tmp/ccTLklKP.s:2072: Error: selected processor does not support `dsb' in ARM mode
In file included from usleep.c:56:0:
xtime_l.h:87:9: note: #pragma message: For the sleep routines, Global timer is being used
 #pragma message ("For the sleep routines, Global timer is being used")
         ^~~~~~~
In file included from xil_sleeptimer.c:51:0:
xtime_l.h:87:9: note: #pragma message: For the sleep routines, Global timer is being used
 #pragma message ("For the sleep routines, Global timer is being used")
         ^~~~~~~
boot.S: Assembler messages:
boot.S:410: Error: selected processor does not support `fmrx r1,FPEXC' in ARM mode
boot.S:412: Error: selected processor does not support `fmxr FPEXC,r1' in ARM mode
Makefile:64: recipe for target 'standalone_libs' failed
make[1]: *** [standalone_libs] Error 1
make[1]: Leaving directory '/home/waqar/workspace/pavel-demin/red-pitaya-notes/tmp/bare_metal_test.cpu1/app_cpu1_bsp/ps7_cortexa9_1/libsrc/standalone_v6_8/src'
Makefile:29: recipe for target 'ps7_cortexa9_1/libsrc/standalone_v6_8/src/make.libs' failed
make: *** [ps7_cortexa9_1/libsrc/standalone_v6_8/src/make.libs] Error 2
make: Leaving directory '/home/waqar/workspace/pavel-demin/red-pitaya-notes/tmp/bare_metal_test.cpu1/app_cpu1_bsp'

```