pavel-demin/red-pitaya-notes #848: Trying mcpha application with square wave

ssooffiiaannee, 2019-05-25:
### Description of the setup:

 * Device: Red Pitaya
 * SD card image: Provided by RedPitaya https://redpitaya.readthedocs.io/en/latest/quickStart/SDcard/SDcard.html
 * Application: MCPHA
 * Other relevant information:

### Description of the problem:
I'm trying to check my understanding of the pha program, I generated a 100 HZ square wave, 4 Volts pip-to-pic, plugged it to the input of the red pitaya, If my understanding is correct the pha core will acquire the digital value of the hight level and the low level, of the square wave and make a substraction, so its the same thing as taking 4 and converting it to its digital value, knowing that the RP has 14 bits adc, from minus to plus 20 volts, so we get (4/40)*16384 = 1638 channel number, but the histogram shows a pic at around 2500.
 I put the baseline to auto, Knowing that the pic is growing at the frequency of the square wave, which is logic, we have a rising edge 100 times per second.

### Steps to reproduce the problem:

 1.Generate square wave with 4 volts VPP, 100 HZ
 2.Plug the signal to IN1 of the RP
 3.Open MCPHA application

pavel-demin, 2019-05-25:
It's normal that MCPHA application doesn't work with square pulses.

The MCPHA application is designed to work with Gaussian-shaped pulses. It also works with exponentially rising and decaying pulses if the rise time is about a few tens of nanoseconds and the decay time is about a few tens of microseconds.

closed 2019-05-25 by pavel-demin

pavel-demin, 2019-05-25:
BTW. A screenshot of the MCPHA client program showing the pulse shape attached to this issue would make much more sense than the screenshot showing almost nothing that you attached to the [previous issue](https://github.com/pavel-demin/red-pitaya-notes/issues/846).