pavel-demin/red-pitaya-notes #850: mcpha baseline characteristics

Hamam-S, 2019-05-26:
### Description of the setup:

 * Device: Red Pitaya 125-14
 * SD card image: Image from red pitaya website
 * Application: multichannel pulse height analyzer  (mcpha)
 * Other relevant information: 

### Description of the problem:
Dear Pavel,
         Really impressed by the quality of your work.
          
         I was wandering what is the difference between setting and not setting the baseline to auto. I tried both and I see huge spectra differences ( Sorry for not being able to put any pictures, I'll do it as soon as possible).

But what is the difference in a more abstract way, when I read the pha verilog program(link : https://github.com/pavel-demin/red-pitaya notes/blob/master/cores/axis_pulse_height_analyzer_v1_0/axis_pulse_height_analyzer.v) , my hypothesis about the difference is this : 
1-  In non-auto baseline, the program takes the maximum of an impulse and substracts the value of the baseline.
2- In auto baseline, the program looks for the minimum value before any rising edge, and puts it as a baseline.

Note : I put the min threshold and max threshold to 0 and 16384 respectively. 
Here is a picture representing my assumptions.

[image]

I supposed that 0 baseline value represents 0 volts, Is it true ?

pavel-demin, 2019-05-26:
Your description of the two baseline modes is correct. Earlier today,  I've updated the notes about the MCPHA application with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/a9294f0347f84b05731b0378c11158aae63a1049) and added a description of the  baseline modes that is similar to yours.

If I'm not mistaken, there is some voltage offset at the Red Pitaya inputs. So, 0 V at the board's input corresponds to a non-zero value at the ADC output.

I think that for the pulse height analyzer to work properly, the pulses should be non-overlapping. In its current state, the MPCHA application doesn't have any pulse shaping module. However, it should be relatively easy to add a simple pulse shaping module that would delay, scale and subtract the delayed signal from the non-delayed signal as it's for example described in [Digital signal processing for BGO detectors](http://dx.doi.org/10.1016/0168-9002(93)91105-V) and in [Digital gamma-ray spectroscopy based on FPGA technology](https://doi.org/10.1016/S0168-9002(01)01925-8).

Hamam-S, 2019-05-26:
Thank you for this huge clarification. 

> If I'm not mistaken, there is some voltage offset at the Red Pitaya inputs. So, 0 V at the board's input corresponds to a non-zero value at the ADC output.

This may explain the spectrum with manual baseline  = 0  below,  since 0 v gives non-zero digital value, we may never read 0 digital input, so we end up with the huge peak of 10000000 counts when we set a manual baseline of 0, and which also explains why there is not count at channel inferior to approximately 85.

[image]

The second hypothesis is that there is noise, so the perturbations that prevent the voltage to go to zero when there is no gamma source adjacent to the detector. So there is constant lecture of non-zero digital inputs that appear on the histograms.

pavel-demin, 2019-05-26:
With the baseline set to 0 and with the long exponentially decaying tails of the pulses, there should be a lot of noise over all the spectrum.

Here is a screenshot showing the noise on the exponentially decaying edge of a pulse:

[image]

All the small peaks are counted if the baseline is set to 0.

So, ideally, the pulses should be shaped before analyzing them with the MCPHA application. Otherwise, only the automatic baseline subtraction should be used.

closed 2019-05-28 by pavel-demin