pavel-demin/red-pitaya-notes #851: Alpine SD Card not defaulting to 192.168.1.100

sixtemesseven, 2019-05-26:
Alpine SD Card Image not Defaulting to 192.168.1.100

I (unziped and) copied the files of the alpine sd image to my original rp sd card, moved the start.sh file from mcpha to the top level and started the rp. It seems to work - LED0 blinks, ETH LED's are active.

I use a D-Link DUB-E100 ver.D ETH adapter (Network 9) manually configured to:
192.168.1.5 assuming it will default back to 192.168.1.100 and directly wired it to the rp. However, I can not ping the rp or connect to it via the browser at all (via rp or via the ip...).
According to nmap, there are no other hosts up on 192.168.1.x

[image]

Is there a way to change the ip or get a log of some kind out of the rp?

pavel-demin, 2019-05-26:
Thanks for the screenshot and for the details about your network setup.

Before switching to 192.168.1.100, the DHCP client on the Red Pitaya board tries to obtain an IP address via DHCP. It switches to 192.168.1.100 after 20 seconds.

Have you waited for 20 seconds before trying to ping the board?

> Is there a way to change the ip or get a log of some kind out of the rp?

I'd suggest to check the boot process and the IP address of the Red Pitaya board using the USB/serial console:

https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html

It's possible to use [PuTTY ](https://www.putty.org) to work with the console.

The commands to check the IP address are
```
ifconfig
```
or
```
ip addr
```

pavel-demin, 2019-05-26:
BTW. If you set the IP address of the USB/Ethernet to 192.168.1.5, please make sure that other network interfaces (Ethernet or Wi-Fi) don't use IP addresses from the 192.168.1.x subnet.

sixtemesseven, 2019-05-26:
No other devices are on that eth adapter and windows will warn you if you have multiple network cards with the same sub address...

Tried to go the serial route. I can connect, but I see a Zynq> user in the console so something is wrong here.

[image]

sixtemesseven, 2019-05-26:
Without SD there is nothing. Plugging SD into running device gives me this: 

[image]

sixtemesseven, 2019-05-26:
I have not modified the uEnv.txt file on the sd (drive E) and it seems to be identical to the one supplied by the zip file...
[image]

pavel-demin, 2019-05-26:
> Plugging SD into running device gives me this:

I'm not sure if I understand what exactly you're doing.

The correct sequence should be
- switch off the Red Pitaya board
- plug SD card
- connect a USB cable to the console port on the Red Pitaya board and to the PC
- connect PuTTY to COM7
- switch on the Red Pitaya board

sixtemesseven, 2019-05-27:
Hmm, tied with a new SD card, works fine. 

[image]

closed 2019-05-27 by sixtemesseven