pavel-demin/red-pitaya-notes #865: mcpha FIR filters

Hamam-S, 2019-06-04:
### Description of the setup:

 * Device:  Red Pitaya 125-14
 * SD card image:  Image from red pitaya website
 * Application: multichannel pulse height analyzer (mcpha)
 * Other relevant information:

### Description of the problem:
Dear Pavel, 
     I noted that in your MCPHA project, you used a CIC filter before the FIR filter, when I googled the CIC filter, I found that we usually use a FIR filter for compensation, but the problem, is that looking at the coefficients you used for the FIR filter, it does not correspond to a compensation filter. 

Here is the frequency response of the FIR filter used in mcpha : 

[image]

and here is the frequency response of a CIC filter with the compensation filter (this is not the frequency response of the CIC filter used in the project, since I don't know how to plot that filter with matlab), this one is from the xilinx doc https://www.xilinx.com/support/documentation/ip_documentation/cic_compiler/v4_0/pg140-cic-compiler.pdf: 

[image]

So my question is what is the purpose of using the FIR filters ?

pavel-demin, 2019-06-04:
If you think that there is a problem in the MCPHA application, then please provide more convincing evidence than a link to the documentation of the CIC compiler that I use in this application and that I of course read before using the CIC compiler.

> when I googled the CIC filter, I found that we usually use a FIR filter for compensation

The linked document doesn't say "usually". Here is what is written on page 12:

> The increased droop might not be acceptable in some applications.

I don't think that "might" and "some" should be interpreted as "usually".

Filter characteristics depend on the requirements of a particular project.

As far as I know, the compensation filters are used in the applications that require a flat frequency response. For example, the [SDR receiver application](http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/) requires a flat frequency response and it has a FIR filter that compensates for the drop in the CIC frequency response. You can see more details in the [digital down-converter section](http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/#digital-down-converter) of the SDR receiver notes.

In case of the MCPHA application, I think that a flat frequency response isn't required. Moreover, a FIR filter that compensates for the drop in the CIC frequency response would increase the amount of ringing and overshoot. So, the pulses would be more difficult to analyze.

> So my question is what is the purpose of using the FIR filters ?

I believe that both the [first-order differentiator](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_pulse_height_analyzer_v1_0/axis_pulse_height_analyzer.v#L48) and the [pulse height calculation](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_pulse_height_analyzer_v1_0/axis_pulse_height_analyzer.v#L49) in the [pulse height analyzer module](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_pulse_height_analyzer_v1_0/axis_pulse_height_analyzer.v) perform better if noisy high frequencies are filtered out of the input signal. So, the role of the FIR filter in the MCPHA application is to smooth the signal and to suppress high frequencies.

BTW. I got the impression that you didn't find my notes about the MCPHA application. They can be found at the following link:

http://pavel-demin.github.io/red-pitaya-notes/mcpha

Hamam-S, 2019-06-05:
I can not access the sciencedirect research papers.
Thank you !!

closed 2019-06-05 by Hamam-S