pavel-demin/red-pitaya-notes #866: Red pataya DAC and DDS compiler

motagiyash, 2019-06-11:
Bonjour Pavel demin, thank you for all your work on red pitaya.

I am working on the project where i am reusing your DAC ip and a DDS compiler, i am able to generate required frequency.

but now i want the Output wave only for 200 ns after every 1000 ns wait period.

I have 20 mhz freq output from dds. so i want like after every 1000 ns or some delay, the 200 ns the dds output should be on oscilloscope.

I tried couple of options but no luck.

is there any options you suggest that i should try.

Thank you very much.

pavel-demin, 2019-06-11:
I think that it's quite similar to the TX part of the pulsed NMR project:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/pulsed_nmr/tx.tcl

Here is [a link to an issue comment with a screenshot](https://github.com/pavel-demin/red-pitaya-notes/issues/503#issuecomment-326407642) showing the output signal.

motagiyash, 2019-06-12:
Thank you very much for the guidence and reply.

The design is bit complex as i am a beginner university student, but from what i understand, You are streaming the phase increment and phase offset values to DDS compiler ?

If yes do you have any documentation where i can read more about the theory of phase increment.

I am thinking of developing my own AXI stream IP to input the values. 

I have also attached the image of what i am tring to achive.

I know its a basic question but i am still little confused.

Thank you very much for your time.

[image]

pavel-demin, 2019-06-12:
I'm afraid you ask too much.

If I share the source code of my applications for Red Pitaya and answer some questions about them, it doesn't mean that I'm interested in doing someone's homework for them.

I believe that finding answers in datasheets and user guides is an essential part of this kind of projects.

closed 2019-06-12 by pavel-demin

motagiyash, 2019-06-12:
Yeah I understand, Thank you for your project work on RP. I has been really helpful.

Have a good day.