pavel-demin/red-pitaya-notes #868: Red pitaya DAC issue

motagiyash, 2019-06-24:


pavel-demin, 2019-06-24:
> Bonjour Mr Pavel,

I'm surprised to see "Mr" used in front of a given name. Is it how we use "Mr" these days in France? If you don't mind, then I'd much appreciate something simpler like "Hi Pavel".

> I am running DDS on 100 MHz and DAC 125 MHz.

For simplicity, I'd suggest to use the same clock source for all the modules.

> Could you guide me where the problem might be.

Looking at the Vivado screenshot, I'd say that the DAC interface module isn't clocked correctly. I'd suggest to use exactly the same clock sources for the DAC interface module as I use in all my projects. I think that [dac_test](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/dac_test) is the most basic project showing how to correctly use  the DAC interface module.

closed 2019-06-25 by pavel-demin

motagiyash, 2019-06-28:
Thank you Mr pavel, for the help and the project work I have solved the issue buy properly modifying PINC and POFF and rsync. 

Have a good day.