pavel-demin/red-pitaya-notes #877: Ethernet controller on zynq

z7010student, 2019-07-11:
Hi Pavel,

I have been reading the zynq TRM chapter 16 about Gigabit Ethernet controller on Zynq PS.

I see on your SDR_transciver project you use where the idea is to transmit ethernet packets to pc.

I wanted to know where is the code(To learn with an example), where you actually program the registers of Ethernet controllers.(Or am i missing somthing)

Thank you and have a good day.

pavel-demin, 2019-07-11:
All my applications for the Red Pitaya board run under Linux. The communication with the Ethernet controller is handled by the corresponding Linux driver:
https://github.com/Xilinx/linux-xlnx/tree/master/drivers/net/ethernet/cadence

closed 2019-07-11 by pavel-demin

z7010student, 2019-07-11:
Thanks pavel,

Its bit complex for me to understand, but i will start working on it now.

My idea was to transmit the data from the FIFO in PL to the TX buffers directly, using the help of DMA avaliable on ethernet controller module.

Do you think it is possibe, by the driver.

Thank you again.

pavel-demin, 2019-07-11:
No idea.

z7010student, 2019-07-23:
Hello Pavel
I am learning to build my own linux from scratch for redpitaya where i can configure the device tree.
I am following Xilinx guidelines "https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841973/Build+U-Boot"
But i am stuck at the board config file step.
They have stated that config file for every board is avaliable in include/config.
eg: zenboard,zc702 etc

How can i get hold of red pitaya file.

Thank you very much your work is really helpful for newbies like me.