pavel-demin/red-pitaya-notes #878: wspr receiver not starting

vronp, 2019-07-14:
### Description of the setup:

 * Device: RP 14 bit
 * SD card image: red-pitaya-alpine-3.9-armv7-20190527.zip
 * Application: wspr transreceiver (receive only)
 * Other relevant information:

### Description of the problem:

Hi, the sdr app does not start.  I followed instructions for moving the start.sh script not to mention the fact that I cannot even start it manually.

I did have some initial issues understanding that Alpine uses the "lbu" command to commit changes to the sd card.  Once I learned of this, I was able to make permanent changes to things like motd.  I still can't change the hostname however which is bothersome and makes me wonder if there is something fundamental I am missing here.

I cannot find any types of logs that help me debug this.  I am looking in /var/log and also at dmesg output.  I'm really scratching my head at this point and would appreciate any help.

thanks,
Dave
n9qnz

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2019-07-15:
Hi Dave,

>  I followed instructions for moving the start.sh script not to mention the fact that I cannot even start it manually.

If you move `start.sh`, then you loose the possibility to start the app manually or via the web interface.

The `start.sh` script should be **copied** to the topmost directory on the SD card. If you do it after booting the Red Pitaya board, then the path is `/media/mmcblk0p1` and the commands are
```
rw
cp /media/mmcblk0p1/apps/sdr_transceiver_wspr/app/start.sh /media/mmcblk0p1/
ro
```

>  I still can't change the hostname

The hostname is defined in `/etc/hostname` and it's overwritten by [/etc/conf.d/hostname](https://github.com/pavel-demin/red-pitaya-notes/blob/master/alpine/etc/conf.d/hostname) at boot. I think that removing [/etc/conf.d/hostname](https://github.com/pavel-demin/red-pitaya-notes/blob/master/alpine/etc/conf.d/hostname) should solve the problem.

Best regards,

Pavel

vronp, 2019-07-16:
Hi Pavel,

Thank you for the reply.

Just to be clear.  You specify "rw" and then "ro" commands.  Is this
putting the system in those modes with the lbu command?  I am fairly
familiar with Linux but not Alpine Linux and am new to working with SD
cards and linux.

So, if you could, please give me a hint about the rw and ro (I assume
read/write and read only)

thanks,
Dave

pavel-demin, 2019-07-16:
`rw` and `ro` are two aliases defined in [`.profile`](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/alpine.sh#L147-L150):
alias rw='mount -o rw,remount /media/mmcblk0p1'
alias ro='mount -o ro,remount /media/mmcblk0p1'

These aliases are for the SD card mounted read-only as `/media/mmcblk0p1`.

The `lbu` commands are for the overlay file system mounted as `/`. This file system is stored in `red-pitaya.apkovl.tar.gz`.

Mounting SD card read-only isn't a bug it's a feature. It allows the following:
- switching power off at any moment without damaging the file system on the SD card
- extending SD card life

vronp, 2019-07-16:
Ok, understood.  I believe I should probably start over at this point with
this new knowledge.

Thanks again.

closed 2019-07-24 by pavel-demin