pavel-demin/red-pitaya-notes #880: SD Card does not boot

n4zrpete, 2019-07-27:
### Description of the setup:

 * Device: Red Pitaya 122.8-16
 * SD card image: https://github.com/pavel-demin/red-pitaya-notes/releases/download/20190527/red-pitaya-alpine-3.9-armv7-20190527.zip
 * Application: SDR receiver compatible with Hermes
 * Other relevant information: Tested on two different computers using Windows 10

### Description of the problem :I used Windows 10 to unzip the image above to a Windows folder, then used Windows copy to copy the contents of the image to a FAT32-formatted micro SD card.  The resulting SD card looks OK - in the sense that all of the files and folders appear to be same as the original folder.  As suggested, I copied the start.sh file to the top of the SD card.  However, when I insert the SD card the Red Pitaya does not boot.  

### Steps to reproduce the problem:

 1.  See above
 2.
 3.

pavel-demin, 2019-07-27:
The SD card images from this repository aren't compatible with STEMlab 122.88-16 SDR.

Please use the SD card images from

https://github.com/pavel-demin/stemlab-sdr-notes/releases

closed 2019-07-27 by pavel-demin

n4zrpete, 2019-07-27:
Ahah - I had assumed that because of the later release date that the one 
I downloaded was compatible.  Thanks!

73, Pete N4ZR
Check out the Reverse Beacon Network
at <http://reversebeacon.net>, now
spotting RTTY activity worldwide.
For spots, please use your favorite
"retail" DX cluster.