pavel-demin/red-pitaya-notes #881: Wideband SDR transceiver ( Red-Pitaya connection problem)

uiuatiq, 2019-07-30:
### Description of the setup: My host pc is running windows 10 and guest VM install with debian 9.

 * Device: Red Pitaya
 * SD card image: Pavel Demin customized SD card image zip file for alpine linux.
 * Application: Wideband SDR transceiver
 * Other relevant information:  I believe that my sd card is ok and copy the red-pitaya-alpine-3.9-armv7-20190527 is successfully. By ethernet cable my laptop and my red pitaya board connected directly.

### Description of the problem: First of all i am very new in Red Pitaya and my English is also weak. So please don't mind if i say something dumb. I have installed development machine on my computer as suggested here : http://fbalakirev.github.io/red-pitaya-notes/development-machine/

N:B: But i did not install vivado sdk as i think i dont need it for the project that i intend to implement.

 I have 8Gb sd card which i first write the stable OS for red pitaya and by ssh connection it is giving web interface as well. Now my aim is to check out the projects that Pavel Demin supplying by sd card zip file ( red-pitaya-alpine) . I have followed the same instructions is given in this link: http://fbalakirev.github.io/red-pitaya-notes/sdr-transceiver-wide/

I must mentioned that I have format the sd card first and copy the sd card zip file and extract it also. Even i copy the start.sh file to the top directory of the file but no pop up file coming during booting(as it suppose to come from the description i see) . Now, when I have put the sd card to red pitaya it seems not connection at all. I have also wait like 2 min to get started. I have have no longer web interface with red-pitaya also. Now only green led on and other led is totally off stage, however previously when i was putting stable os on sd card red led was blinking, green was on as well as blue was on and web interface was working.

However, I need to say that i am using ssh and x11 forwarding (ximing) and putty is for ssh.
This ssh connection is working and i am able to login through root/changeme. so it is not problem with ssh. 

My pc is DHCP enabled.

Even though the red pitaya seems not connected at all But I still try to run the gnuradio part by command line.

cd red-pitaya-notes/projects/sdr_transceiver_wide/gnuradio
export GRC_BLOCKS_PATH=.
gnuradio-companion trx_wide_template.grc

and the gnu run ( I think this is because I clone it as well)but whenever I run it this produce a lot of error. Even I try to change some things with .grc file but still no luck.

What is going wrong here, I do not understand. Am I missing something serious?  is there any way that i can use stable os with sd card and also copy the sd card zip file by Pavel Demin in the same card?

### Steps to reproduce the problem:

 1. sd card ready to use by copy the red-pitaya-alpine-3.9-armv7-20190527
 2. By ethernet cable my laptop and my red pitaya board connected.
 3. May be I am making some mistakes , please help someone.
[image]
[image]
[image]
[image]
[image]
[image]

pavel-demin, 2019-07-30:
Thanks for all the information. However, I think that I still need more information to understand your problem.

Let's start with the information about your device.

Where are currently at least three models of the Red Pitaya board: 125-14, 125-10 and 122.88-16. Which one are you using?

uiuatiq, 2019-07-30:
Hello Pavel, Thank you so much for your quick response. 

Unfortunately I do not have the red pitaya with the box thats its came.  But from google search i think this the model of 122.88-16. I just compare all model images with my one and for 122.88-16 it is like the same image. I am not sure.

Or may be you can kindly tell me how to find the model name to be sure.

Thank you again.

pavel-demin, 2019-07-30:
Please post a picture of both sides of your board.

uiuatiq, 2019-07-30:
Hello Pavel, 

Yes , here I have attached my red pitaya board picture of the both side. 

Please have a look. Thank you so much.

[image]
[image]

pavel-demin, 2019-07-30:
Thanks for the pictures. Since the board doesn't have input transformers and it has a micro USB console port, it's 125-14.

Now we are sure that your board is compatible with the SD card images from this repository.

I think that the next step is to understand what happens during the boot.

I'd suggest to check the boot process of the Red Pitaya board using the USB/serial console:

https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html

It's possible to use [PuTTY ](https://www.putty.org) to work with the console.

If the boot process works, then you should see the following:
[image]

uiuatiq, 2019-07-30:
Hello Pavel, 
Thank you so much for you reply.

I need to say that I am using the Ethernet crossover cable to connect red pitaya with my laptop directly.  Is to Is it ok to use that thernet crossover cable? I am bit confused here in this point. 

Another point is I never seen the Debug console part. I just saw the link that you gave and I do not get clear idea about it. But as much as i understand i am doing. I used a power supply separately, and the crossover cable as well as I also trying the  USB to microUSB cable connection between laptop and the red pitaya board (as link followed, and i think this is for serial commutation establishment).

Then in putty  forward x11 for ssh, then in session select the serial to com7 then click open. But nothing is happening, because it seems putty is loading. I wait 5 and more minutes to see what happen next but unfortunately nothing is happening.

Thank you again for you excellent support. 

[image]
[image]
[image]
[image]

pavel-demin, 2019-07-30:
Thanks for the screenshots. The  Red Pitaya documentation recommends setting the speed to 115200. Could you try replace 9600 with 115200.

If it doesn't help, then I think that something is wrong with the SD card and I'd suggest the following:

 - take a freshly unpacked micro SD card (or FAT32 format a used micro SD card)
 - unpack the red-pitaya-alpine-3.9-armv7-20190527.zip file to the micro SD card
 - insert the micro SD card in Red Pitaya and connect the power

uiuatiq, 2019-07-31:
Dear Pavel,
First of all thank you for your reply. 
I think there is some problem with my work. Because I have did everything you have suggested except using a new sd card. 
I have set up the speed 115200, but it is also not working as same situation continue. Then I format my sd 8GB sd card with taking FAT32. Furthermore, i did unpack the red-pitaya-alpine-3.9-armv7-20190527.zip file to the micro SD card and as well as insert the micro SD card in Red Pitaya and connect the power.
Unfortunately nothing is happening. 
So, now i have to try for last chance to use a new sd card. And sorry for late reply because i was taking time to do this steps again and again. 

I have a question to ask you though. When I insert the sd card (copy with red-pitaya-alpine-3.9-armv7-20190527) to the Red Pitaya only GREEN Led is on and all other Led is off. Is it normal situation?

Thank you so much for your patient. 

[image]
[image]
[image]

pavel-demin, 2019-07-31:
The problem is clearly seen on the last screenshot.

Only the files and the directories from the .zip file should be copied to the micro SD card:
- double click on red-pitaya-alpine-3.9-armv7-20190527.zip
- select all files and directories (Ctrl+A)
- copy all the selected files and directories (Ctrl+C)
- go to the micro SD card
- past the files and directories (Ctrl+P)

No additional directories should be created.

uiuatiq, 2019-07-31:
Hello Pavel,

Thank you so much. Yes, I think it is working now. 
I login by root/changeme. One more thing to ask you that how can i run wideband SDR Transceiver now? Would you like to give me the idea? I ask you this beacuse i follow this line : ' Optionally, to start the application automatically at boot time, copy its start.sh file from apps/sdr_transceiver_wide to the topmost directory on the SD card' but nothing happening.

I am so grateful to you for your help.

[image]
[image]
[image]
[image]

pavel-demin, 2019-07-31:
> One more thing to ask you that how can i run wideband SDR Transceiver now?

The easiest would be to start it via the web interface.

> copy its start.sh file from apps/sdr_transceiver_wide to the topmost directory on the SD card' but nothing happening.

If the first yellow LED (LED0) stops blinking after the boot, then the application is started.

You can also check if the `sdr-transceiver-wide` process is running using the `ps aux` command.

uiuatiq, 2019-07-31:
Hello Pavel, 
Thank you for your quick reply. 
'If the first yellow LED (LED0) stops blinking after the boot, then the application is started.', yes after booting LED= stop blinking.
'The easiest would be to start it via the web interface.' yes, I put http://rp-f0xxxx.local/ where xxxx is the mac address written on RP. But web interface is not coming.
I run the command ps aux and it seems like  sdr-transceiver-wide is running on the system.
I know that i am asking many questions from you but what should i do now?
Thank you for your help.

[image]
[image]

pavel-demin, 2019-07-31:
> I run the command ps aux and it seems like sdr-transceiver-wide is running on the system.

So, the autostart functionality is working.

> what should i do now?

Check the network connection between the PC and the Red Pitaya board.

What are the IP addresses of the PC and the Red Pitaya board?

uiuatiq, 2019-07-31:
Hello Pavel,
I know my RP mac address and I did not give any static ip to red pitaya. But let me try now, My VM's ip is 192.168.152.1 and my host PC ip is 172.31.1.178. As my PC and VM are both set to DHCP then I guess RP will get a ip address automatically? may be I am wrong. And my laptop and RP connected through crossover cable. Another point is from note i read ''The wired interface is by default configured to request an IP address via DHCP. If no IP address is provided by a DHCP server, then the wired interface falls back to a static IP address 192.168.1.100.'' Therefore, I ping this ip address from my PC and seems like 192.168.1.100 is my RP ip address.

Thank you so much and looking forward to get your feedback.

[image]
[image]
[image]

pavel-demin, 2019-07-31:
Could you please check the IP address of the Red Pitaya board via the USB/serial console using the `ifconfig` command?

> As my PC and VM are both set to DHCP then I guess RP will get a ip address automatically?

If the network configuration is the same as you described in the first comment:

> By ethernet cable my laptop and my red pitaya board connected directly.

then you should manually assign an IP address to you PC. Something like 192.168.1.2.

uiuatiq, 2019-07-31:
Hello Pavel,
Thank you so much for your reply.
'Could you please check the IP address of the Red Pitaya board via the USB/serial console using the ifconfig command?'
yes, I did check that. I found that exact the mac address of RP, ip address corresponding to 192.168.1.100. So, I try to enter this ip to web and find nothing(no web interface came).
Then I have to do the static ip assigning.  I did that and seems like it is working. In web I put ip of RP as showing. It is directed to the all apps notes. What should be the next step? Until now is it seems normal?

Thank you again for your great patient. 

[image]
[image]
[image]

pavel-demin, 2019-07-31:
> What should be the next step?

I think that everything is OK now and GNU Radio should be able to communicate with your Red Pitaya board.

pavel-demin, 2019-07-31:
BTW. The web page show on your last screenshot can be used to run all the preinstalled application. Just click on one of the links and the corresponding application will be started.

dish-eme-sdr, 2019-07-31:
I have bee reading this thread since it started.
I want to thank you for your time and devotion to this project.
This level of support is extremely rare for ANY project and is VERY much appreciated !!!
Even though I have been playing around with the RedPitaya for several years, I have picked up some helpful hints (by taking the time to read this thread) for when things don't go as planned.
Hopefully, if I keep playing around with this project long enough I will be able to make some meaningful contributions...

Again

A great big

!!!=< T-H-A-N-K   Y-O-U >=!!!
.

pavel-demin, 2019-07-31:
Thank you @dish-eme-sdr for your kind words!

uiuatiq, 2019-07-31:
Hello,
I honestly feel there is so much with RP that I must study. As a beginner (start playing 8 days ago)all i can say is Pavel is superbly helping me.
Pavel has been extremely supportive all around. I have a very little knowledge about RP and as a students I can not say enough but thanks to Pavel. In this thread I have get so much help. When I was first start this thread, I was thinking may i will not get any response. But, Pavel replied and everything become easier. I even feel silly about my questions but Pavel help in such way that i never expected. I feel joy to get help from someone like Pavel.

Again I have to ask Pavel a silly question though. I have checked that LED Blinking Project working fine. For Wide band SDR I have still problem. This is because the .grc file is not opening while serial comm. was made by Putty.
As far as I understand from the link of projects note that I have to install gnuradio on RP, so before I did this when I was connected through SSH.  So, as .grc is not openning on serial comm., then i shift the putty connection to SSH and write gnuradio-companion and .grc file open but RP block are missing.

'Connect an antenna to the IN1 connector on the Red Pitaya board' I don't have one right now but can i see the graph as in normal simulation gnuradio program?

Thank you very much for answering my all questions so kindly. 

[image]
[image]
[image]

pavel-demin, 2019-07-31:
> As far as I understand from the link of projects note that I have to install gnuradio on RP,

GNU Radio should be installed on a computer, **not on Red Pitaya**.

Since you have a screenshot of GNU Radio Companion running, then you have already GNU Radio installed.

So, you have already everything installed.

> but RP block are missing

I think that you forgot to set the `GRC_BLOCKS_PATH` variable before starting GNU Radio Companion.

Here are the correct commands to start GNU Radio Companion:

```
cd red-pitaya-notes/projects/sdr_transceiver_wide/gnuradio
export GRC_BLOCKS_PATH=.
gnuradio-companion trx_wide_template.grc
```

uiuatiq, 2019-08-01:
Hello Pavel,
Thank you so much for your help. Finally It is working. I am going to change the .grc file to play with it.
I have been so lucky to get continues help from you. 

[image]

closed 2019-08-01 by pavel-demin