pavel-demin/red-pitaya-notes #886: FT4 on the new STEMlab

F1EYG, 2019-09-15:
Hello Pavel
I just be back on PSKreporter with my new STEMLab :-)) the result is terrific 
(STEMLab for FT8 on all band/OldPitaya for CW on 8 band)
Do you plan to ad FT4 ?
With the power of the new STEMLab do you think is possible to run concurrent HPSDR
AND FT8 applications at the same time ?

pavel-demin, 2019-09-24:
Hello Sylvain,

I see that your call sign is now at the very first position of the Top Monitors lists on the PSK Reporter Statistics page. Congratulations with your very well performing setup!

I don't have any plans about FT4 right now but it may change in the future.

The FT8 decoder running on the Red Pitaya board consumes nearly all the available CPU resources. So, I don't think that running the HPSDR and FT8 applications at the same time on the same board would work. However, with some modifications of the FT8 application, it's possible to move the FT8 decoding to a separate more powerful computer. I think that with the FT8 decoding  running on a separate computer, the new STEMLab 122.88-16 SDR board should have enough resources to run simultaneously an eight-band HPSDR receiver and an eight-band FT8 receiver.

Best regards,

Pavel

closed 2019-09-24 by pavel-demin

F1EYG, 2019-09-24:
Hello tks for your answer 😀 i have test to use pytaya like just receiver and running sparksdr on a w7 pc 😅 just for fun but i need to put all on a box 😢 
73

dj1an, 2019-12-09:
Hi,
just tried to decode the redpi ft8_*.c2 files on another machine with the "original" jt9 decoder but the file can't be read.
Any hint how to convert the c2 files to wav or a compatible format jt9 can read?
Or is it necessary to patch the decoder?

73

azwirko, 2020-01-10:
> Hi,
> just tried to decode the redpi ft8_*.c2 files on another machine with the "original" jt9 decoder but the file can't be read.
> Any hint how to convert the c2 files to a compatible format jt9 can read?
> Or is it necessary to patch the decoder?
> 
> 73

I had originally toyed with the idea of using the WSJT-X 'jt9' command line wave file decoder on the RedPitaya ARM processor to decode FT8 and more recently FT4 signals.  For FT8 I'd had some success, but found the decoding quite a bit slower than Pavel's optimized re-write of the 'ft8d' code.  Anyhow 'jt9' does work on the RP and here is what I did to decode FT4 using the existing .c2 files produced by Pavel's FPGA code.  

First for FT4 I had to change a frequency line in the write-c2-file.cfg file to an FT4 frequency, for example I used 14081500 

I re-compiled WSJT-X for ARMv7 - this was actually done on an RPi running Alpine Linux 3.9 and then copied the 'jt9' binary and the support libraries to the RP

I compiled HA7ILM's CSDR code for ARMv7 and copied it and associated libraries to the RP

I needed to install 'sox' for use to convert raw I/Q to WAV IQ file format used by 'jt9'

After that I ran Pavel's FT8 .bit code on the FPGA and manually copied the .c2 files from /dev/shm to the SDcard for later processing

The later processing to convert and then decode FT4 on the RP included the following set of commands:

`dd ibs=512 count=473 skip=1892 if=ft8_7_14081500_1_200105_2037.c2 | csdr shift_math_cc 0.32 | csdr bandpass_fir_fft_cc 0 1.0 0.05 | csdr realpart_cf | csdr agc_ff | csdr limit_ff | csdr convert_f_s16 > file.raw`

`sox -t raw -r 4000 -b 16 -e signed-integer -c 1 file.raw -r 12000 000000_123415.wav`

`time ./jt9 -5 -d 3 000000_123415.wav`

The resultant FT4 decode and associated timing on the RP was as follows:

> 001234  -4 -0.1  427 +  CQ RU AE0JM EM10
> 001234  -7 -0.0  643 +  CQ RU K2AK DM41
> 001234   8  0.1 1372 +  VE3KI VE6WQ R 549 AB
> 001234  11 -0.1 1773 +  W6SX N3FKZ 569 TX
> 001234   7 -0.1 1013 +  KM4HI KB5PGY R 569 TX
> 001234  -7 -0.1 1306 +  KI3F KA0KVW 569 CO
> 001234   8 -0.1 1738 +  CQ RU AA8R EL97
> 001234 -15  0.0  923 +  CQ RU W0RAA DM78
> 001234   8 -0.0 1053 +  CQ RU N4BP EL96
> <DecodeFinished>   0   9
> real    0m 5.89s
> user    0m 5.77s
> sys     0m 0.11s

73

andyz - K1RA

dj1an, 2020-01-10:
Hello Andy,
Thanks for your response,
in the meantime i managed to write a gnuradio flowgraph which does demod and slices the c2 files in 15s chunks. 
The complete decoding happens on a second computer, so the redpi is now degraded to only record files and send them over the net.

73 de DJ1AN, Jan

azwirko, 2020-01-10:
Jan
  Very good - I've thought of that too but really like the idea of doing everything of the RP.  I've put a request into the WSJT-X developers if they could produce an 'ft4d' command line program similar to that of their 'ft8d' version.  If they can, that would be a good starting point to begin what Pavel had done in optimizing and compiling a stand alone program to handle FT4 .c2 files on the RP directly.  I'll post more if there is any progress on that front.

73

andyz - K1RA

azwirko, 2020-05-09:
Jan

I've recently been working on something that you may be interested in.  See my post here:

https://groups.io/g/RBN-OPS/message/4358

73

andyz - K1RA