pavel-demin/red-pitaya-notes #887: Scan engine under active development?

joverbee, 2019-10-01:
hi Pavel,

I was wondering if there is active delopment on the scan engine? I am interested in using it for an SEM setup but want to make sure I am starting at the most up to date point. I have an old RPitaya but could upgrade to the newer 16 bit version if needed. Ideally I would like to be able to define in a lookup table the x,y dac points that are visited sequentially at the rate of an external clock input to allow for alternative scan strategies (e.g. 1us dwell time could be enough). I am not sure this could be done fast enough and if the lookup table memory would allow for e.g. 1024x1024x2x2 byte space.
I basically want to figure out if this is the right point to start or maybe you know of related developments which would be more relevant.

Kind regards,
Jo

pavel-demin, 2019-10-02:
Hi Jo,

Thanks for your interest in the [scanning system](http://pavel-demin.github.io/red-pitaya-notes/scanner/) application. It was built for a specific project and worked as expected. So, no additional development was required nor planned.

The current version doesn't have an external clock input. All the clock and trigger signals are generated by the Red Pitaya board. So, the code should be modified to enable this functionality.

I don't think that the new 16-bit Red Pitaya board is suitable for this project. It has AC coupled outputs and inputs.

Best regards,

Pavel

pavel-demin, 2019-10-02:
I've looked through the code and I think the additional functionality that you project requires can be easily added to the [scanning system](http://pavel-demin.github.io/red-pitaya-notes/scanner/) application. I'll try to do some tests and I'll post the results in this issue.

joverbee, 2019-10-02:
Wonderful! thanks a lot for looking into this.

pavel-demin, 2019-11-03:
I've added an external clock input and a lookup table memory with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/15ff967bd4568ed6aa3d5f75d689c7b179ee48e2).

An external clock should be connected to the pin DIO0_N of the [extension connector E1](http://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extent.html#extension-connector-e1).

If you want to give this new version a try, here is a link to a SD card image containing the new version of the scanner application:

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AABWc_KLUM79pdrGg8JD9PoQa/scanner/red-pitaya-alpine-3.9-armv7-20191103-scanner.zip?dl=1

joverbee, 2019-11-04:
Great! Will try to test out soon.

joverbee, 2019-11-08:
[image]
It works! (after some trouble learning how to set things up and ordering a new RP 14 bit board). On the picture we use my old RP 10 bit as pulse generator to sync the scanner. 
We are planning to extend the user interface a bit to allow for setting up the number of pixels in a scan, the 'coordinate' pattern loading from file, stopping the scan while scanning etc. Looking at the code these seems fairly obvious extensions that we can do that remain on the python side only.
When we have hooked it up to a real SEM, we will send you a nice picture.
Thank you very much.

pavel-demin, 2019-11-20:
Thanks for the test and for the picture.

There is now a release with the updated scanning system:
https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20191120

joverbee, 2019-12-10:
We finally got it hooked up to our old Jeol SEM and everyhting seems to run smooth. We tested up to 1us dwell time (1Mhz sync input) and this works well as long as the python interface is not updating while receiving buffers. We will now focus on the python side and provide possiblity to tune aspect ratio and keep the widest output range independent of the nr of pixels scanned (voltage swing of Rpitaya is a bit limiting, but dont feel like adding an output stage at this point). We also plan to implement a button to load a scan pattern.
[image]

Arpit-Dave, 2020-01-08:
Hi Pavel,

Firstly, thank you very much for developing this Scanning system. And thank you to Jo for opening this issue, along with attaching pictures that are helpful.

I have been working on developing a LabVIEW program to control the system for the 2D scanning Galvo mirror using Red Pitaya's STEMlab 125.14 board. After investing a lot of time and effort, I am finding it increasingly difficult to work on LabVIEW for any Red Pitaya development. From the scouring of Red Pitaya's forum and having gotten the not-so-helpful responses from the Red Pitaya, specifically pertaining to the LabVIEW development, this Scanning system based on Python comes as a great respite for me. I have been working toward developing a standalone python code, but since I have already spent so much time and effort on LabVIEW, I needed a step forward. So, thank you again for sharing your development.

I have followed the steps for the Windows environment given on http://pavel-demin.github.io/red-pitaya-notes/scanner/, but I am facing a couple of issues. After having copy-pasted the extracted contents of the customized SD card image, and after inserting the SD card and powering up, my Red Pitaya board's LED0 blinks for about 32 times and then stops. Also, only the blue and green led lights are on, and not the red and yellow ones which indicates that something is wrong. I had formatted the SD card before copy-pasting the files onto it. Can you please help me out here?

I did install the recommended Python 3.4 and then PyQT 5.5 from the cited links. Then, downloaded scanner.py and scanner.ui. But then, nothing happens after running C:\Python34\pythonw.exe scanner.py to run the control program. Do I need to download and save both the files at a specific location? What wrong am I doing here?

If I get the Scanning system to work, it will be of huge help and a leap forward for me. Hope to hear from you soon.

Kind regards,
Arpit

pavel-demin, 2020-01-08:
Hi Arpit,

From your description, the Red Pitaya part seems to be OK. The LEDs behave exactly as they should.

Have you installed numpy and matplotlib? I'd suggest to run `C:\Python34\python.exe scanner.py` (without `w`) and to check the error messages.

Best regards,

Pavel

pavel-demin, 2020-01-09:
Looks like the latest matplotlib version doesn't support Python 3.4 anymore. So the installation command should be
```
C:\Python34\Scripts\pip.exe install numpy==1.15.4 matplotlib==2.2.4
```

Arpit-Dave, 2020-01-09:
Thank you, Pavel, for your prompt response. I had installed numpy and matplotlib. With revised version of matplotlib, as you suggested, I did install it. I ran the command, `C:\Python34>pythonw.exe scanner.py` first and like it was previously, nothing happened.

I ran the command `C:\Python34>python.exe scanner.py` and following is the error message that I am getting:

```
Traceback (most recent call last):
  File "scanner.py", line 37, in <module>
    Ui_Scanner, QMainWindow = loadUiType('scanner.ui')
  File "C:\Python34\lib\site-packages\PyQt5\uic\__init__.py", line 194, in loadUiType
    winfo = compiler.UICompiler().compileUi(uifile, code_string, from_imports, resource_suffix)
  File "C:\Python34\lib\site-packages\PyQt5\uic\Compiler\compiler.py", line 111, in compileUi
    w = self.parse(input_stream, resource_suffix)
  File "C:\Python34\lib\site-packages\PyQt5\uic\uiparser.py", line 972, in parse
    document = parse(filename)
  File "C:\Python34\lib\xml\etree\ElementTree.py", line 1186, in parse
    tree.parse(source, parser)
  File "C:\Python34\lib\xml\etree\ElementTree.py", line 587, in parse
    source = open(source, "rb")
FileNotFoundError: [Errno 2] No such file or directory: 'scanner.ui'
```

I am not sure what wrong did I do here. Do I need to save the files scanner.py and scanner.ui at a particular location?

Thank you for helping me. Grateful for your precious time.

Kind regards,
Arpit

pavel-demin, 2020-01-09:
> Do I need to save the files scanner.py and scanner.ui at a particular location?

scanner.ui should be in the directory where your run scanner.py. So I'd suggest to copy both files to `C:\Python34`.

Arpit-Dave, 2020-01-09:
> scanner.ui should be in the directory where your run scanner.py. So I'd suggest to copy both files to `C:\Python34`.

Yes, that's what I had done. I wanted to get the confirmation on the same. But the error message still persists upon running `C:\Python34>python.exe scanner.py`.

To be precise, I had the file contents extracted from the version 'red-pitaya-alpine-3.9-armv7-20191120-scanner.zip' https://github.com/pavel-demin/red-pitaya-notes/releases/download/20191120/red-pitaya-alpine-3.9-armv7-20191120-scanner.zip and copy-pasted on to the SD card. Just to check with another version that's cited here, I tried, after formatting the SD card, copy-pasting the version 'red-pitaya-alpine-3.9-armv7-20191103-scanner.zip' from the link cited above, https://www.dropbox.com/sh/5fy49wae6xwxa8a/AABWc_KLUM79pdrGg8JD9PoQa/scanner/red-pitaya-alpine-3.9-armv7-20191103-scanner.zip?dl=1. But the error message is the same.

Kind regards,
Arpit

pavel-demin, 2020-01-09:
The python error message is unrelated to the SD card installation. If the LEDs blink as you described previously, your SD card installation is OK.

I just tried to reproduce your problem with python but everything worked without any error message. Here's what I did:
- installed [Python 3.4](https://www.python.org/ftp/python/3.4.4/python-3.4.4.msi) and [PyQt 5.5](https://sourceforge.net/projects/pyqt/files/PyQt5/PyQt-5.5.1/PyQt5-5.5.1-gpl-Py3.4-Qt5.5.1-x32.exe/download)
- downloaded [scanner.py](https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/scanner/client/scanner.py) and [scanner.ui](https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/scanner/client/scanner.ui) directly to `C:\Python34`
- started a command prompt using the `cmd.exe` program and ran the following commands
```
C:
cd \Python34

python.exe -m pip install --upgrade pip
python.exe -m pip install numpy==1.15.4 matplotlib==2.2.4

python.exe scanner.py
```

pavel-demin, 2020-01-09:
Since you mentioned the feedback loop and Thorlabs Galvo Mirror Systems in the [post](https://forum.redpitaya.com/viewtopic.php?t=23656&p=30824#p30824) on the Red Pitaya forum, I think that I should mention that the scanning system application discussed in this issue is an open loop system.

Arpit-Dave, 2020-01-09:
> Since you mentioned the feedback loop and Thorlabs Galvo Mirror Systems in the [post](https://forum.redpitaya.com/viewtopic.php?t=23656&p=30824#p30824) on the Red Pitaya forum, I think that I should mention that the scanning system application discussed in this issue is an open loop system.

Yes, you are right, Pavel. I should probably clarify furthermore in the post. I meant to convey the feedback loop of the entire control system.

Arpit-Dave, 2020-01-09:
> The python error message is unrelated to the SD card installation. If the LEDs blink as you described previously, your SD card installation is OK.
>
> I just tried to reproduce your problem with python [...]

Thank you for taking the time out to cross-check the error that I was getting. I am happy to let you know that that issue has been resolved. It was the issue from my side actually. When I was downloading and saving the scanner.ui file to `C:\Python34` directory, it was, by default, getting saved as a text file, instead of .ui file. Once I corrected that, the GUI window did open as mentioned by you.

I have made all the hardware connections and now it is the time to get my image. I am really excited. 

There is only one thing now that whenever I am writing the IP address of the Red Pitaya board, there pops up an error message box saying "Error: Connection timed out." Now, ever since adding the customized SD card image contents, I haven't been able to see the IP address or the corresponding MAC address of the Red Pitaya board from Command Prompt and using `arp -a`. The IP address that I was trying to write in the text box of the Scanner GUI was from the memory of having made the IP address of the board static and having used it many number times on different PCs. So, the IP address is definitely correct, and exactly what I have using for a while. 

Since I can't see the IP address in the command prompt, do you have a suggestion on how to overcome this, hopefully, the last issue? Since you mentioned that the LEDs are behaving as they should, do you think that not getting to see the IP address of the board is in line with its inherent behavior of the code? I have actually only been able to take a glance through the acquisition part of the code. I will definitely check more from my side, but, your input will be really helpful.

Thank you!

pavel-demin, 2020-01-09:
I'd suggest to check the boot process and the IP address of the Red Pitaya board using the USB/serial console:

https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html

It's possible to use [PuTTY ](https://www.putty.org) to work with the console.

The commands to check the IP address are
```
ifconfig
```
or
```
ip addr
```

More information about the network configuration, login, password, etc can be found at [this link](http://pavel-demin.github.io/red-pitaya-notes/alpine).

pavel-demin, 2020-01-09:
In the current version, a clock/trigger signal should be connected to the pin DIO0_N of the [extension connector E1](http://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extent.html#extension-connector-e1). The internally generated pulses are available from the pin DIO0_P. So for a test without an external clock/trigger generator the pin DIO0_N should be connected to the pin DIO0_P.

Arpit-Dave, 2020-01-10:
> In the current version, a clock/trigger signal should be connected to the pin DIO0_N of the [extension connector [...]

Yes, thank you.

Arpit-Dave, 2020-01-10:
> I'd suggest to check the boot process and the IP address of the Red Pitaya board using the USB/serial console:
>
> https://redpitaya.readthedocs.io/en/latest/developerGuide/os/console/console.html [...]

Hi Pavel,

Thank you for the directions. I did also check the network configuration webpage that you provided. Here's what I did:

1. I disconnected the ethernet cable and connected the Red Pitaya board directly to the PC as mentioned in the console.
2. Following the console furthermore, I did install the FTD driver and checked that the new COM port (default, COM3) was added to my Device Manager list.
3. I, then, opened PuTTY, selected Serial connection type and entered COM3 as serial line and then clicked on the Open button. After that, a blank PuTTY terminal window opens and if I type something, only three horizontal-lines-kind-of characters get written. I have previously used PuTTY for Red Pitaya to log in through the IP address. 
4. I, then, connected the ethernet cable again and tried to use PuTTY. Since I didn't get to the point to use the commands `ifconfig` or `ip addr`, I tried my known IP address but the "Connection timed out" error came up. I did also try to use the static IP 192.168.1.100, but, the same error popped up.
5. Just to check whether my IP address information was correct, I took out the SD card and formatted it to reinstall the SD card image provided by Red Pitaya. I found that the IP address was the same as I was trying to use. I also noticed that it was with my old Red Pitaya board, I had made its IP address as static. But with this current board, it is still with the DHCP. 
6. Now, I have once again, freshly installed the SD card image provided by you and have been following steps 1 to 3 but the result is the same.

Can you please guide me furthermore if you have any other suggestions? Am I missing any step here? My apologies for taking your time but I am slightly on the newer side when it comes to resolving this particular issue.

Thanks again for your patience and all the help.

Kind regards,
Arpit

pavel-demin, 2020-01-10:
> After that, a blank PuTTY terminal window opens and if I type something, only three horizontal-lines-kind-of characters get written.

Looks like you forgot to correctly set the serial port speed. It should be set to 115200.

Since your last few comments aren't directly related to the scanning system, please open a new issue and move these comments/questions there. Once it's done, I'll remove all our unrelated comments/questions/answers from this issue. I prefer to keep this issue for discussions that are directly related to the scanning system.

Arpit-Dave, 2020-01-10:
> Looks like you forgot to correctly set the serial port speed. It should be set to 115200.
>
> Since your last few comments aren't directly related to the scanning system, please open a new issue and [...]

I have moved the relevant conversations from here to the new issue https://github.com/pavel-demin/red-pitaya-notes/issues/922.

closed 2020-05-02 by pavel-demin