pavel-demin/red-pitaya-notes #898: Linux development from scratch.

z7010student, 2019-11-06:
Hello Pavel,

I am trying to build linux first time for my GPIO project.

I have been following Xilinx guide (https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841961/Zynq+Linux) for build of uboot and kernel. I would appretiate if you guide me through this one issue. 

As to build Uboot its stated that we need, .dts and _defconfig file. I went through your repo, the files you have in you repo -> patches .

the question is that, are thise files your application specific? so should i create or modifi `those?

` CC      examples/standalone/stubs.o
  LD      examples/standalone/libstubs.o
  CC      examples/standalone/hello_world.o
  LD      examples/standalone/hello_world
  OBJCOPY examples/standalone/hello_world.srec
  OBJCOPY examples/standalone/hello_world.bin
  LDS     u-boot.lds
  LD      u-boot
arch/arm/mach-zynq/built-in.o:(.data.fpga+0x14): undefined reference to `zynq_op'
cmd/built-in.o: In function `eeprom_rw_block':
/home/yash/linux_yash/u-boot-xlnx/cmd/eeprom.c:149: undefined reference to `i2c_read'
/home/yash/linux_yash/u-boot-xlnx/cmd/eeprom.c:151: undefined reference to `i2c_write'
cmd/built-in.o: In function `do_i2c_reset':
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:1942: undefined reference to `i2c_init'
cmd/built-in.o: In function `do_i2c_probe':
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:987: undefined reference to `i2c_probe'
cmd/built-in.o: In function `do_i2c_crc':
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:768: undefined reference to `i2c_read'
cmd/built-in.o: In function `mod_i2c_mem':
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:863: undefined reference to `i2c_read'
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:911: undefined reference to `i2c_write'
cmd/built-in.o: In function `do_i2c_md':
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:583: undefined reference to `i2c_read'
cmd/built-in.o: In function `do_i2c_read':
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:336: undefined reference to `i2c_read'
cmd/built-in.o: In function `do_i2c_mw':
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:681: undefined reference to `i2c_write'
cmd/built-in.o: In function `do_i2c_write':
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:419: undefined reference to `i2c_write'
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:405: undefined reference to `i2c_write'
cmd/built-in.o: In function `do_i2c_loop':
/home/yash/linux_yash/u-boot-xlnx/cmd/i2c.c:1080: undefined reference to `i2c_read'
cmd/built-in.o: In function `do_thor_down':
/home/yash/linux_yash/u-boot-xlnx/cmd/thordown.c:42: undefined reference to `thor_init'
/home/yash/linux_yash/u-boot-xlnx/cmd/thordown.c:49: undefined reference to `thor_handle'
common/built-in.o: In function `jumptable_init':
/home/yash/linux_yash/u-boot-xlnx/common/exports.c:30: undefined reference to `i2c_write'
/home/yash/linux_yash/u-boot-xlnx/common/exports.c:30: undefined reference to `i2c_read'
Makefile:1495: recipe for target 'u-boot' failed
make: *** [u-boot] Error 1
`
Thank you for your time pavel.

pavel-demin, 2019-11-06:
> are thise files your application specific?

I think that they are quite generic.

closed 2019-11-06 by pavel-demin

z7010student, 2019-11-06:
> > are thise files your application specific?
> 
> I think that they are quite generic.

Thanks Pavel.

do you have any pointers according to the errors?
I have used the .dts,config.h and defconfig from your project.

pavel-demin, 2019-11-06:
If you think that there is a problem, then please provide a list of commands to reproduce it.

As far as I know, my scripts and patches work when they are used with the commands from my notes.

Most of the commands can be found in the LED blinker notes:

http://pavel-demin.github.io/red-pitaya-notes/led-blinker/

Some more details can be found in the slides of my presentation at Club Vivado 2016. They are linked from the LED blinker notes. Here is a direct link:

http://pavel-demin.github.io/red-pitaya-notes/led-blinker/
https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACl--BhQvcNgjeQLRaiX9dha/ClubVivado2016_Pavel_Demin.pdf?dl=1