pavel-demin/red-pitaya-notes #905: External PA and LNA control/monitoring (Temp, current, output power, swr, lna gain)

satfan52, 2019-11-22:
Hello,

It seems the so called  "Hermes" protocol allows to monitor these parameters.  QUISK for instance allows to display the following information provided that the Hermes hardware is equipped with the adequate sensors and controls. See picture below

[image]

Of course the redpitaya board doens't have an internal PA or internal LNA like most Hermes board do. I also know it supports Metis which I understand is a little different than Hermes.  Yet some GPIO pins and an I2C addresses are available on the RP and could possibly be used to connect external sensors or controls devices to manage/monitor external PA and LNA within the limits/capabilities of the Hermes/Metis protocols.

=> So I wonder if would be possible to modify the software of the RP so it is possible to connect external sensors / control device supported by the so called Hermes/Metis protocols

pavel-demin, 2019-11-22:
It's already implemented.

The power inputs are described in the [RF, GPIO and XADC connections](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr/#rf-gpio-and-xadc-connections) of the notes. Here is a copy of this description:
- slow analog inputs can be used for the forward ([Analog input 0](http://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extent.html#extension-connector-e2)) and reverse ([Analog input 1](http://redpitaya.readthedocs.io/en/latest/developerGuide/125-14/extent.html#extension-connector-e2)) power measurement

The analog inputs 2 and 3 are used for the current and voltage monitoring.

closed 2019-11-22 by pavel-demin

satfan52, 2019-11-22:
Thanks, never really used those analog inputs, to the point I even forgot their existence on the RP.
 So this addresses
- swr (analog input 0 and 1)
- current (analog input 2)
but what about 
- temperature sensing , looks like an input, is it input 3? 
-  the RfLNA  control , this looks like an output not an input?
- the RfAGC,  same this looks like an output not input ?

pavel-demin, 2019-11-22:
The RED Pitaya board has only four analog inputs and all four are already used by the SDR transceiver compatible with HPSDR.

Here is link to the code that puts the values read from the ADC to the data packets that are sent to the SDR program:
https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_transceiver_hpsdr/server/sdr-transceiver-hpsdr.c#L1263-L1283

The values read from the ADC are put to the positions that are called AIN1, AIN2, AIN3 and AIN4  in [HPSDR - USB Data Protocol](https://github.com/TAPR/OpenHPSDR-SVN/raw/master/Documentation/USB_protocol_V1.58.doc).

In PowerSDR mRX PS, these values are shown as forward power, reverse power, current and voltage.

satfan52, 2019-11-23:
Thank you, very interesting I will study this carefully. I note that none of the analog outputs are used currently. I also note that if I really needed, I could create an additional input using a free address on the I2C interface and using external I2C based AD Converter such as the MCP4725